open62541/open62541 #3089: Casting Extension Object data to UA_HistoryData* fails

luibass92, 2019-09-07:
Hello everyone,

I'm playing around with HistoryReadRequest and HistoryReadResponse and I am monitoring everything with wireshark.
I am currently sending correct HistoryReadRequest and receive also correct HistoryReadResponse (checked with wireshark indeed) to/from a dummy server which simply has 2 historized nodes and the values are changed manually thorugh UaExpert.

Now I would like to unpack the HistoryReadResponse to extract the HistoryData in my code, the point is that this type contains an ExtensionObject and I think I am missing something.

The incriminated part of code is the following

`UA_HistoryData* l_historyData = reinterpret_cast<UA_HistoryData*>(l_historyReadResponse.results[i].historyData.content.decoded.data);`

where l_historyReadResponse should contain the right values according to wireshark but, after this assignement the values are not completely well copied to l_historyData (surprisingly the dataValuesSize is correct).

Can someone help me?
Thanks,

Luigi

jpfr, 2019-09-10:
The content of `historyData.content.decoded.type` should tell which exact type is wrapped in the ExtensionObject.

Depending on your configuration it is also possible that the DataType definitions are not built into the client SDK. Then, the ExtensionObject contains the binary encoded string that was received.

luibass92, 2019-09-11:
Hi jpfrand thanks for your reply,

> The content of `historyData.content.decoded.type` should tell which exact type is wrapped in the ExtensionObject.

yes I can see that `typeIndex = UA_TYPES_HISTORYDATA`, so it is what I am expectiong.
Moreover I can see also that `historyData.content.encoding = UA_EXTENSIONOBJECT_DECODED`.
That is why I did a "brutal" cast on `void* data`  to `UA_HistoryData*`, but still I am not obtaining the correct data.

jpfr, 2019-09-11:
> Maybe could be something wrong with the UA_HistoryData_copy() funciton?

Very unlikely. The function is auto-generated fromt the XML definition of the Datatype.

https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/tools/schema/Opc.Ua.Types.bsd#L2304

But I cannot exclude a problem "upstream" where the HistoryData is first generated.

luibass92, 2019-09-16:
> Very unlikely. The function is auto-generated fromt the XML definition of the Datatype. [...]

Hi jpfr,

just to clarify, there are no problems in the library, I had probably some data with the field `hasValue` set to false and I did not noticed it, so when I tried to de-reference that void pointer (nullptr), obviously the program crashed.
Now I did several test with a check on the field `hasValue` and it worked properly.

Anyway thanks for your support, I close this issue :)

Luigi

closed 2019-09-16 by luibass92