open62541/open62541 #3090: Is there an existing reference implementation for persisting history data?

Jonas-Bartkowski, 2019-09-08:
I was wondering if anyone has already implemented a flat-file and/or sql(ite) persistence version of [ua_history_data_backend_memory.c](https://github.com/open62541/open62541/blob/master/plugins/historydata/ua_history_data_backend_memory.c). I'm asking because I need one and wanted to check that I'm not reinventing the wheel.

So far I had no luck finding one, however I saw [this](https://github.com/open62541/open62541/issues/1735) discussion, which tells me an example on persistence was at least planned.

jpfr, 2019-09-10:
The current historizing backend was developed by Basyskom.
@basyskom-meerkoetter @peter-rustler-basyskom 

It would be great to see different backends.
Sqlite is a good candidate. In additon, there are a few more options specifically for time series data.

**Ordered key-value stores**
- LMDB: https://en.wikipedia.org/wiki/Lightning_Memory-Mapped_Database
- RocksDB: https://github.com/facebook/rocksdb

**Specialized File Formats**
- HDF5: https://de.wikipedia.org/wiki/Hierarchical_Data_Format
- Parquet: https://en.wikipedia.org/wiki/Apache_Parquet

Since the backends should be exchangeable, I don't have a particular preference for any technology.
But beware that some changes to the backend API might become necessary to properly generalize.

As always, changes to the public library API should be discussed early to avoid duplciate work.

Persistence for historizing would be very cool to have.
Looking forward to seeing more from your end!

andreasebner, 2025-02-04:
There is currently no known work being done on persistent history backends. If there are any updates, we will announce them in the community call.