open62541/open62541 #3091: Default DataSetFieldContentMask in PubSub

GalaxyGorilla, 2019-09-09:
Hey,

I'm currently working on PubSub within our own OPC UA Stack (written in Erlang). I'd love to use your example code regarding PubSub to guide me further through the implementation and of course for compatibility checks.

I wonder about a detail regarding the _DataSetFieldContentMask_ which is used to configure the encoding of DataSet fields. Possible encodings are RawData, Variant or DataValue, depending on configured timestamp and error handling. Which default _DataSetFieldContentMask_ do you use (on current master)? From the examples this is not obvious.

jpfr, 2019-09-10:
This is set in the config of the DataSetWriter.
See the logic here: https://github.com/open62541/open62541/blob/master/src/pubsub/ua_pubsub.c#L1737
This is then evaluated in the encoding layer.

I recommend to use only support Variant and DataValue.
With those, the receiver can decode the message without out-of-band communication of the content field types and order.

Cool, Erlang is nice.
Are you going for a high-concurrency situation or what is the use-case for Erlang?

GalaxyGorilla, 2019-09-11:
@jpfr Many thanks, I will go for Variant or DataValue :). I will check out the example receiver!

Well, at this point of time our OPCUA framework is rather a research project but we have our own IoT focused board (checkout www.grisp.org) which runs the Erlang VM "bare metal" in combination with RTEMS to get pretty good real-time characteristics. So in the future we plan to use OPCUA of course together with our board but the usecase is certainly not restricted to that :).

entone, 2020-03-29:
@GalaxyGorilla did you ever make any progress with the erlang library? Big fan of the Grisp board!

GalaxyGorilla, 2020-03-29:
@entone Yes, we have an Erlang based OPC-UA Stack with a limited feature set, but unfortunately it is not open source (yet).

teccheck, 2025-02-13:
The code paths around this were overhauled quite a lot, so I'll close this issue. Interesting project though :)

closed 2025-02-13 by teccheck