open62541/open62541 #3094: I want to build open62541 in Borland C++ Builder 6. I can see the instructions to do the same in Visual Studio. how can I build this in C++ Builder 6? Also, what are the other options to build it without using cmake? I am usinng 64 bit Windows 10 and C++ Builder 6.

mc-gearup, 2019-09-10:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-09-10:
See here: http://docwiki.embarcadero.com/RADStudio/Rio/en/Using_CMake_with_C%2B%2B_Builder

Just reopen if questions remain that are particular to open62541.

closed 2019-09-10 by jpfr

mc-gearup, 2019-09-11:
The link you have provided is for using CMake with C++ Builder. My question is regarding building open62541 in C++ Builder 6.  Also, I don't want to use CMake , and I can't find any leads on how to do that. Is it possible to build the open62541 in C++ Builder 6?

schroeder-, 2019-09-11:
C++ Builder 6 is an ancient compiler from 2002! I think your best chance is to download a prebuild dll from https://open62541.org/ and use the dll from your application.