open62541/open62541 #3099: open62541.h file missing after build in Windows 10

mc-gearup, 2019-09-11:
Hi, I am trying to Build a Simple server using open62541 as described in this link : https://open62541.org/doc/current/tutorial_server_firststeps.html.

I am using **Borland C++ Builder 6** in **Windows 10 (x64 processor)** and the newest version of open62541. I cloned the repository to my PC and build it using CMake as described in this link : [Building open62541](https://open62541.org/doc/current/tutorial_server_firststeps.html).

but, I am not able to find the open62541.h file in any of the folder (which is metioned as a mandatory file in the link which shows the Simple server example). I did not get any error while building open 62541 using cmake (i tried both from command prompt and CMake GUI).

I am using: 

- CMake 3.10.0

- Python 2.7

- Borland C++ Builder 6

Can someone please throw some light on how to proceed further?

Thankyou

mc-gearup, 2019-09-11:
[image]

mc-gearup, 2019-09-13:
@crafti5 so far what i have done is, I cloned the open62541 and build it using CMAKE with C++ Builder (using the Borland Makefiles generator). But, even after a successful build, the 'src_generated' folder was not creating the files in it. So, I used MS Visual studio 2017 and build a project and this generated the .h files in 'src_generated' folder. I copied these files to my C++ Builder project , but I am getting errors in mainly 'log.h' and 'types.h' folder like:

[bcc32 error] log.h (101): E2140 Declaration not allowed here
   Full parser context
     myServer.c (13): #include C: \ Users \ Development \ open62541 \ include \ open62541 / plugin / log_stdout.h
     log_stdout.h (10): #include C: \ Users \ Development \ open62541 \ include \ open62541 / plugin / log.h

jpfr, 2019-10-22:
Do you compile as C or as C++?
Maybe C-mode expects ANSI C89 and not C99.
You should be able to compile the code as C++98 as well.

andreasebner, 2025-02-04:
The question seems to be answered. If there are new questions, feel free to create a new issue.

closed 2025-02-04 by andreasebner