open62541/open62541 #3102: Unexpected EventType in ModelChangeEvent

Heimetli, 2019-09-13:
I have a fairly recent version of open62541 (two weeks or so) built with 
cmake -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON . on Linux (Debian)

After that, I took tutorial_server_events.c from the examples directory and built it.

It runs without any problem, but UaExpert complains at every event: "Unexpected EventType in ModelChangeEvent", and no events are shown in the event window.

aDogCalledSpot, 2019-09-16:
I can produce the error message in UaExpert but events should be showing up. Have you added the server to the monitored nodes ("Configuration" in the Event View in UaExpert).

Heimetli, 2019-09-16:
You are right: if configured correctly, the events will show up.

But the messages about "Unexpected EventType" are still worrying me, because I don't know how the  users client will react to this.

Is there a way to turn off the "Alarms and Conditions are not supported yet!" messages on the server side?

ccvca, 2020-04-07:
I've investigated this error a little bit further now.

- UaExpert subscribes to BaseModelChangeEventType on the server-node (always, even when the event view is not used)
- When an event in send, this subscription is triggerd, as there seems to be an error in the filtering (I don't know if this happens in all cases)
- As the EventType is not a BaseModelChangeEventType, UaExpert shows a warning.

juangburgos, 2020-04-20:
@ccvca I also get a lot of these:

[image]

Is is something not being done correctly on the open62541 server side or on the UaExpert client side?

mlang-de, 2020-04-20:
I am not sure if this is the problem, but you respect the behavior for the property "NodeVersion"?
Is a ModelChangeEvent required property.

[image]

ccvca, 2020-04-20:
> Is is something not being done correctly on the open62541 server side or on the UaExpert client side?

If I get it right, it's an issue in open62541. UaExpert create a subscription for ModelChange events (filter is set, so only ModeChangeEvent (of subtypes?) should be received). But the open62541 seems to ignore this filter and sends any event to this subscription. So UaExpert complains, that the event does not have the expected (ModelChangeEvent) type.

mlang-de, 2020-04-20:
Our implementation works without UaExpert warnings.
[image]

Check the attached wireshark record. 
[Wireshark_ModelChangeEvent.zip](https://github.com/open62541/open62541/files/4504359/Wireshark_ModelChangeEvent.zip)

I guess the AffectedType field is not correct. "i=10637" is the OffNormalAlarmType. But there shall be the Type of the ParentNode were the model changes.

juangburgos, 2020-04-20:
@ccvca makes sense, because I get those warning for totally unrelated events. Like when an alarm is triggered.

mlang-de, 2020-04-20:
Furthermore the implementation triggers some other NOT base information model events and conditions without UaExpert warnings.

juangburgos, 2020-04-20:
@mlang-de are you using the current master?

mlang-de, 2020-04-20:
Unfortunately no. Some modified v1.0 version.
I am not sure if the was some rework in the event module.

juangburgos, 2020-04-20:
I think there has been some new code added to support Alarms and Conditions, maybe?

mlang-de, 2020-04-20:
Do you use the master from today? In PR https://github.com/open62541/open62541/pull/3540 was a bug fix in A&C.

ccvca, 2020-04-21:
> 
> 
> > Is is something not being done correctly on the open62541 server side or on the UaExpert client side?
> 
> If I get it right, it's an issue in open62541. UaExpert create a subscription for ModelChange events (filter is set, so only ModeChangeEvent (of subtypes?) should be received). But the open62541 seems to ignore this filter and sends any event to this subscription. So UaExpert complains, that the event does not have the expected (ModelChangeEvent) type.
> 
> PS: It seems like this warning appears each time, an event is triggeed.

I've checked these things only for "normal" events, not for alarms. As Alarms use a separate code for trigering, the error might be different there. When fixing this, we should consider to reuse as much code as possible.

juangburgos, 2020-04-21:
I think the `UA_Server_triggerConditionEvent` internally calls `UA_Server_triggerEvent`:

```c
/* triggers an event only for an enabled condition. The condition list is
 * updated then with the last generated EventId. */
UA_StatusCode
UA_Server_triggerConditionEvent(UA_Server *server, const UA_NodeId condition,
                                const UA_NodeId conditionSource, UA_ByteString *outEventId){
    /* Check if enabled */
    UA_ByteString eventId = UA_BYTESTRING_NULL;
    UA_QualifiedName enabledStateField = UA_QUALIFIEDNAME(0, CONDITION_FIELD_ENABLEDSTATE);
    if(!isTwoStateVariableInTrueState(server, &condition, &enabledStateField)) {
        UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_USERLAND,
                       "Cannot trigger condition event when "
                       CONDITION_FIELD_ENABLEDSTATE"."
                       CONDITION_FIELD_TWOSTATEVARIABLE_ID" is false.");
        return UA_STATUSCODE_BADCONDITIONALREADYDISABLED;
    }

    setIsCallerAC(server, &condition, &conditionSource, true);

    /* Trigger the event for Condition*/
    //Condition Nodes should not be deleted after triggering the event
    UA_StatusCode retval = UA_Server_triggerEvent(server, condition, conditionSource, &eventId, false);
    CONDITION_ASSERT_RETURN_RETVAL(retval, "Triggering condition event failed",);
    
    setIsCallerAC(server, &condition, &conditionSource, false);
    
    /* Update list */
    retval = updateConditionLastEventId(server, &condition, &conditionSource, &eventId);
    if(outEventId)
        *outEventId = eventId;
    else
        UA_ByteString_deleteMembers(&eventId);
    return retval;
}
```

But just to be sure, I am literally running `tutorial_server_events.c` examples from today's **master** branch. every time I trigger the event using the `GenerateEvent` method, I am greeted with the warning:

[image]

juangburgos, 2020-04-21:
> I've investigated this error a little bit further now.
> 
> * UaExpert subscribes to BaseModelChangeEventType on the server-node (always, even when the event view is not used)
> * When an event in send, this subscription is triggerd, as there seems to be an error in the filtering (I don't know if this happens in all cases)
> * As the EventType is not a BaseModelChangeEventType, UaExpert shows a warning.
> 
> Edit:
> Looks like the whereClause is not evaluated while triggering an event (at least I didn't find a refrerence to it in UA_Server_triggerEvent)
> 
> ```c++
> typedef struct {
>     size_t selectClausesSize;
>     UA_SimpleAttributeOperand *selectClauses;
>     UA_ContentFilter whereClause;
> } UA_EventFilter;
> ```

I think @ccvca is right. I don't know much about the internals of how the filtering mechanism should work. I should read the spec. But I get the feeling that the filter is not implemented properly, because is not ignoring the selectClause that asks for `Changes`. The event of the example does not have any `Changes` component.

[image]

sqadasi, 2020-04-21:
@ccvca is right, UaExpert creates a MonitoreItem on the Server Object for BaseModelChangeEvent. The Problem is that MonitoredItem has only 3 filter clauses (for BaseModelChangeEvent) but doesn't include the other clauses from BaseEventType. 
Server Object is the source of all Events, so the server tries to enqueue the triggered event to each MonitoredItem of the Server Object. And here comes the second Problem, which generates the UaExpert warning. The server creates EventNotification with only the 3 filter clauses for the BaseModelChangeEvent and writes wrong data into them, and thus sends a wrong EventNotification to UaExpert.
In my opinion, the server should only enqueue events to the MonitoredItems, which has the correct amount of clauses (I think 19 at least). That could be checked inside the loop:
https://github.com/open62541/open62541/blob/578d521931777181d1a881c8335ca5fafcd66eb2/src/server/ua_subscription_events.c#L530

I guess UaExpert creates that MonitoredItem by defualt to get change notification even when users don't create a MonitoredItem by themselves. However, ModelChangeEvents should be managed differently as @mlang-de stated using NodeVersion Property, and should be triggered explicitly by the server with all clauses inherited from BaseEventType.

ccvca, 2020-04-27:
Just for documntation: This is the relevant Monitored Item Request:
[image]
Everything seems fine to me.

Complete Wireshark Capture:
[UaExpertEventCapture.zip](https://github.com/open62541/open62541/files/4537880/UaExpertEventCapture.zip)

ccvca, 2020-04-27:
Can someone check, if this is fixed by my pull request?
In my Server, the message is gone and all events are still working, but different implementations might cause different results ;)

sqadasi, 2020-04-29:
> 
> 
> Can someone check, if this is fixed by my pull request?
> In my Server, the message is gone and all events are still working, but different implementations might cause different results ;)

@ccvca I tested with Events and A&C Examples and it works.

closed 2020-04-29 by jpfr

juangburgos, 2020-05-05:
> Can someone check, if this is fixed by my pull request?
> In my Server, the message is gone and all events are still working, but different implementations might cause different results ;)

@ccvca thanks for the fix, works like a charm

PR #3572 (2020-04-29): Evaluate whereClaus of event subscriptions for operator OfType
This is a minimalistic implementation of the WhereClause for EventSubscriptions.

Fixes #3102

Limitations: Only UA_FILTEROPERATOR_OFTYPE is evaluted.