open62541/open62541 #3108: Reading/Writing ExtensionObject without knowing the CustomDataType

przluca, 2019-09-16:
## Description
Hello, I'm trying to develop a Client for reading `ExtensionObject` from a Siemens PLC. The `ExtensionObject` is a structured variable made by arrays of `BYTE` variables and arrays of `ExtensionObject`. Each one of the `ExtensionObject` has its own Reference `HasTypeDefinition` to the custom type defined in `Types->DataTypes->OPC Binary->SimaticStructures`.   For instance I have a structured defined as the following:

```
MyStruct
    |--->ID
    |--->MyChildStruct[11] //Array of 11 elements
                |---->ID
                |---->AnotherStruct[10]
                                  |---->ID
```
From UaExpert I see `MyStruct` as an `ExtensionObject` as well as `MyChildStruct` and `AnotherStruct`. 

As far as I understood the standard, when I send a Read Request on the NodeId of `MyStruct`, I should receive all the data from the whole structure encoded in some way. Here's an example of what I did now:

I build a readRequest with the NodeId of `MyStruct` and I call the Service Read
```
readResponse = UA_Client_Service_read(client, readRequest)
```
Then I can check that I received an Extension Object because:
```
readResponse.result[0].value.type->typeId.identifier.numeric == 22 //true
UA_ExtensionObject * eo = UA_ExtensionObject_new();
UA_ExtensionObject_copy((UA_ExtensionObject*)readResponse.result[0].value.data, eo);
```
but now how can I find all the data from here? 

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.0-rc5
 - [ ] Other OPC UA SDKs used (client or server): UaExpert - Siemens PLC
 - [ ] Operating system: CentOS 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-09-16:
Your client needs to have the custom DataType baked in.
https://github.com/open62541/open62541/tree/master/examples/custom_datatype

Otherwise, the ExtensionObject is received but contains a binary (undecoded) blob.

In principle, we could receive the structure metadata at runtime from the server.
But we don't have that feature atm.

przluca, 2019-09-17:
Ok thanks.

przluca, 2019-09-17:
In principle, I need to ask to the Server for custom Data Types. I think I will need the Query Service Set to do this. Am I right? Is it any other way to get the DataType info from the Server?

przluca, 2019-09-18:
Ok I can read NodeId of the customDataType from the Server. But this isn't enough. I have to retrieve the `UA_DataType` information from the Server. How can I achieve this?

przluca, 2019-09-18:
I developed a decoder/encoder for my specific application. The issue is solved.

closed 2019-09-18 by przluca

UlfJohnsson, 2022-09-30:
@przluca 
Im trying to do the exact same thing. I have no problem with decoding expansion-objects, given the structure it's pretty straight forward.
I'm assuming the encoding is the same but reversed?
Can someone please tell me how do I write a value to a extension object variable? It's really driving me nuts!
Im trying this, but no luck, I only get UA_STATUSCODE_BADDECODINGERROR status:

```
// 'data' is a byte array serialized in the way I described above
UA_ExtensionObject *exo = UA_ExtensionObject_new();
UA_ExtensionObject_init(exo);
exo->encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
exo->content.encoded.typeId = UA_NODEID_NUMERIC(2, 4);
UA_ByteString body;
UA_ByteString_init(&body);
UA_ByteString_allocBuffer(&body, data.length());
exo->content.encoded.body.length = data.length();
memcpy(body.data, data.constData(), data.length());
exo->content.encoded.body = body;
		
UA_Variant variant;
UA_Variant_init(&variant);

// convert the data
auto dataType = &UA_TYPES[UA_TYPES_EXTENSIONOBJECT];
UA_Variant_setScalar(&variant, &exo, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]);

UA_Client_writeValueAttribute(m_client, variableNodeId, &variant);
```