open62541/open62541 #3109: Out of tree support of the dependency OPCFoundation/UA-Nodeset

herbrechtsmeier, 2019-09-16:
Add a CMake option to pass a out of tree ua-nodeset directory to the CMake configuration. The directory of the ua-nodeset could be extracted from the existing `UA_FILE_NS0` option or could be added as a new `UA_DIR_NODESET` option.

Pro, 2019-09-16:
There is already such an option, it's called `open62541_NODESET_DIR`:
https://github.com/open62541/open62541/search?q=open62541_NODESET_DIR&unscoped_q=open62541_NODESET_DIR

It is more an internal option rather than a user-provided one, but still you can set it in your CMake options.
It allows you to define the location of the ua-nodeset dir.

Is that the one you are looking for?

herbrechtsmeier, 2019-09-16:
The option `open62541_NODESET_DIR` isn't used inside the main CMakeLists.txt. The path `/deps/ua-nodeset` is hard-coded at different lines. 

I want to move the OPCFoundation UA-Nodeset into its own package to handle it independent of this project.

vladimiroltean, 2019-09-16:
@herbrechtsmeier: Something like this?
https://github.com/open62541/open62541/pull/3110

herbrechtsmeier, 2019-09-17:
@vladimiroltean Thanks for the patch. It would be nice if the package config could use an external uc-nodeset too.

vladimiroltean, 2019-09-17:
What package config?

herbrechtsmeier, 2019-09-17:
I mean the [open62541Config.cmake.in](../blob/master/tools/cmake/open62541Config.cmake.in).

At the moment this project installs files from an other project (uc-nodeset). It would be nice if this could be disabled via an option UC_ENABLE_EMBEDDED_NODESET.

vladimiroltean, 2019-11-12:
I have no idea how to do that, and I have no time to figure that out, sorry.

andreasebner, 2025-02-04:
This issue appears to be out of date. The new options mentioned are part of all recent releases. If there are known issues, feel free to open a new issue.

closed 2025-02-04 by andreasebner