open62541/open62541 #3111: AutomationML file to OPC UA XML for OPC-UA Aggregation Server

RAMI40, 2019-09-17:
Hallo Everyone, 

since this is my first post related to an Issue please excuse me if I am not at the right place or if I have not fully complied to the rules of posting an issue. I actually posted to an existing issue but realized that it is closed. Since I was not sure how to proceed I decided to create an entirely new issue.

I am currently working on my thesis. The thesis involves the implementation of a tool as part of process chain that converts the existing AutomationML models into OPC UA XML that can be further used for OPC UA Aggregation Server.

Is there a possibility for university students to access this tool developed from IOSB (https://www.iosb.fraunhofer.de/servlet/is/35891/) or perhaps a possibilty to acquire license to this tool especially for students.

I would highly appreciate your time and reply. Thank you in advance.

Regards.

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-09-17:
Hey @RAMI40 ,

can you take that up with Ljiljana Stojanovic?
She is the contact person for that topic. And also mentioned on the page.
I can also make an introduction if you send me a private message with your contact details.

The generation of the Nodeset XML is something that happens "before" open62541 comes into play.
Closing this issue for now.

closed 2019-09-17 by jpfr

gustavocainelli, 2021-08-25:
Hello @jpfr ,

I am interested in using the convert (AutomationML / OPC UA XML). Is this tool available for students? Thank you very much

zediia, 2023-10-25:
>Dear @gustavocainelli,
     Hello, did you successfully get the tool? Or have you found something similar? If so, please reply me. Thank you very much