open62541/open62541 #3117: Missing single-file distribution link for release version of open62541 in website

traversaro, 2019-09-19:
## Description
In the website http://open62541.org , under the Download section for the latest release, the link under "Single-file distribution" actually point to regular full source tarballs, where no `open62541.h` or `open62541.c` file can be found.

## Background Information / Reproduction Steps
Go to http://open62541.org, and scroll to the Download section.

Note that single-file distributions for the latest 50 commits are available in https://open62541.org/releases/ , but those files are for testing purposes only and may be removed without any notice.  

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0-rc5 
 - [x] Other OPC UA SDKs used (client or server): NA
 - [x] Operating system: NA
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: NA
 - [x] Wireshark network dump attached: NA
 - [x] Self-contained code example attached: NA
 - [x] Critical issue: No

arpadboda, 2019-09-24:
Big +1 for this, I also need the same.

kaelanfouwels, 2019-10-18:
Has there been any change on this? Would also be much appreciated.

Pro, 2019-10-18:
Just for collecting opinions:
Is there a specific reason why you can't build the amalgamated version yourself?

You would need to adapt the CMake settings anyways to fit your needs.

In general it's recommended to use the non-amalgamated files. The amalgamation should only be used if you have specific build systems, e.g., for micro controllers.

Can you explain a bit more your use-case please?

kaelanfouwels, 2019-10-18:
In my case I’m wrapping the library through CGO, currently linked to the single file 0.3 release, am intending to migrate this to 1.0.

I can happily build it myself if that is the intended release process - It is preferable from a validation perspective to pull and link to an official release.

traversaro, 2019-10-25:
> Can you explain a bit more your use-case please?

In my case, the single file distribution was useful for disseminate open62541 to Visual Studio users that are not familiar with CMake and do not want to deal with the distribution of a `.dll`, and for which a single `.h`/`.c` to include in their Visual Studio solution has a significant lower barrier of adoption then having to learn how to use the CMake GUI, etc, etc. This is an indeed niche use case, so I fully understand if you don't want to offer that as one of the main downloads from the homepage. 

However, the main point of this issue was another: the generic source archive should not be linked under the "Single-file distribution and full source code:" description, as it may be quite misleading, especially because just at the point before there is a link to the "Nightly single-file distributions" that are indeed real single-file distributions. From my point of view, introducing the links to the source archive as "Full source code:" would be sufficient to close this issue.

locka99, 2020-04-16:
I found using the amalgamated files a very easy way to use the code and for the sake of adding a switch to CMake to generate it I think there is benefit in ensuring there is a distribution like that. Personally I used the amalgamated code to write some 3rd party tests against my own OPC UA implementation which is written in Rust. I wasn't looking to inspect the open62541 code, I just needed it to compile and link and the amalgamated code was the easiest way.

teccheck, 2025-02-13:
The recent updates to the site fixed this

closed 2025-02-13 by teccheck