open62541/open62541 #3132: Debian: Should libopen62541-<soname>-dev really install tools at /usr/share/open62541 ?

jrcoding, 2019-09-26:
## Description
The Debian package `libopen62541-<soname>-dev` installs the tools ( ` tools/*.py tools/certs tools/nodeset_compiler tools/schema tools/ua_nodeset`) at location `/usr/share/open62541`.

**Is this really the intention** or should we move this to a separate package (e.g. `open62541-doc` together with the examples and the HTML and PDF documentation)?

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version **1.0**:
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: **Debian Linux**
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-09-26:
I think these scripts belong into the dev package. E.g. if one needs to compile custom NodeSets he needs these scripts.
I am not sure what the best practice is for such a case...

jrcoding, 2019-09-26:
o.k, so the tools really meant for development work, sounds o.k. for me, so let's left it how it is for now...

closed 2019-09-26 by Pro