open62541/open62541 #3137: linking error ("undefined reference") when building master branch

andreucm, 2019-09-26:
## Description

The following linking error ("undefined reference") arises when building with examples and pubsub related code enabled:
```
[ 43%] Building C object examples/CMakeFiles/client_connect_loop.dir/client_connect_loop.c.o
[ 43%] Linking C executable ../bin/examples/client_connect_loop
../bin/libopen62541.so.1.0.0: undefined reference to `UA_PubSubManager_removeRepeatedPubSubCallback'
../bin/libopen62541.so.1.0.0: undefined reference to `UA_PubSubManager_addRepeatedCallback'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/client_connect_loop.dir/build.make:95: recipe for target 'bin/examples/client_connect_loop' failed
make[2]: *** [bin/examples/client_connect_loop] Error 1
CMakeFiles/Makefile2:741: recipe for target 'examples/CMakeFiles/client_connect_loop.dir/all' failed
make[1]: *** [examples/CMakeFiles/client_connect_loop.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

## Background Information / Reproduction Steps
 - Ubuntu18.04
 - gcc 7.4.0
 - cmake 3.10.2
 - opcua62541 branch 1.0 and branch master

I do: 
```
git clone https://github.com/open62541/open62541.git
cd open62541
git submodule update --init --recursive
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo  -DUA_NAMESPACE_ZERO=FULL  -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=ON -DUA_ENABLE_PUBSUB_DELTAFRAMES=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=ON -DUA_BUILD_EXAMPLES=ON ..
make
```

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON 
      -DCMAKE_BUILD_TYPE=RelWithDebInfo 
      -DUA_NAMESPACE_ZERO=FULL 
      -DUA_ENABLE_PUBSUB=ON
      -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=ON
      -DUA_ENABLE_PUBSUB_DELTAFRAMES=ON
      -DUA_ENABLE_PUBSUB_ETH_UADP=ON
      -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON
      -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=ON
      -DUA_BUILD_EXAMPLES=ON
..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0 and master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Any comment or suggestion is welcome. Thanks.

jpfr, 2019-09-28:
Just remove the custom publish handling option.
You won’t need this in most cases.

andreucm, 2019-09-28:
Thanks @jpfr for the reply. 

I toggle the CUSTOM_PUBLISH_HANDLING option, so inside the build directory I do: 

```
rm -rf *
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo  -DUA_NAMESPACE_ZERO=FULL  -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=OFF -DUA_ENABLE_PUBSUB_DELTAFRAMES=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=ON -DUA_BUILD_EXAMPLES=ON ..
make
```

and a similar linking error arises:
```
[ 56%] Linking C executable ../bin/examples/pubsub_subscribe_standalone
/tmp/cc19ojNZ.ltrans0.ltrans.o: In function `subscriberListen':
/home/andreu/dev/z_external/open62541/examples/pubsub/pubsub_subscribe_standalone.c:63: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/andreu/dev/z_external/open62541/examples/pubsub/pubsub_subscribe_standalone.c:101: undefined reference to `UA_NetworkMessage_deleteMembers'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/pubsub_subscribe_standalone.dir/build.make:95: recipe for target 'bin/examples/pubsub_subscribe_standalone' failed
make[2]: *** [bin/examples/pubsub_subscribe_standalone] Error 1
CMakeFiles/Makefile2:1148: recipe for target 'examples/CMakeFiles/pubsub_subscribe_standalone.dir/all' failed
make[1]: *** [examples/CMakeFiles/pubsub_subscribe_standalone.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

Did you manage to build the library with the cmake options I'm using ?

THanks

jpfr, 2019-10-15:
> -DBUILD_SHARED_LIBS=ON
> -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=ON

These two options are not compatible with `-DUA_BUILD_EXAMPLES=ON`.
The first one because the examples use internal API that is not exported to a shared object.
The second one works only with the realtime_publisher example.

So try to disable BUILD_SHARED_LIBS.

dnoliver, 2020-03-11:
Disabling BUILD_SHARED_LIBS works! Thanks.

This sample Dockerfile builds the pubsub samples

```dockerfile
FROM alpine:3.10

RUN apk add --no-cache cmake gcc git g++ musl-dev mbedtls-dev python py-pip make && rm -rf /var/cache/apk/*

WORKDIR /opt

RUN git clone https://github.com/open62541/open62541.git && \
    cd open62541 && \
    git fetch -f --tags && \
    git submodule update --init --recursive

WORKDIR /opt/open62541/build

RUN cmake -DBUILD_SHARED_LIBS=OFF \
        -DCMAKE_BUILD_TYPE=Debug \
        -DUA_BUILD_EXAMPLES=ON \
        -DUA_ENABLE_PUBSUB=ON \
        -DUA_ENABLE_PUBSUB_MQTT=ON \
        -DUA_ENABLE_HARDENING=OFF \
        -DUA_ENABLE_ENCRYPTION=ON \
        -DUA_ENABLE_SUBSCRIPTIONS=ON \
        -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON \
        -DUA_NAMESPACE_ZERO=FULL \
         /opt/open62541 && \
    make -j $(nproc) && \
    make install
```

NoelGraf, 2021-04-29:
I would then close the issue.

closed 2021-04-29 by NoelGraf