open62541/open62541 #3140: Branch pack/1.0 - does version number in debian/changelog increase correctly?

jrcoding, 2019-09-28:
## Description

On 2019-09-26 Travis-CI sets the version number in branch pack/1.0 file CMakeLists.txt to `v1.0` (see 6cea5c36a16baa94190b4254588e5c882153f26b) and on 2019-09-28 Travis-CI sets the version number to `v1.0-1-g4f7fe52f` (see b9939bf29a0d793d5df26cddaf9f08cc2b8e3b70).
As result in the first case the file `debian/changelog` has the version number `1.0` and in the second case `1.0~1~g4f7fe52f`. 

Debian compares `1.0` GREATER `1.0~1~g4f7fe52f`

I'm not sure about the intended numbering scheme but shouldn't the version number increase with `1.0~1~g4f7fe52f`?

If so, I think the correct number should be like  `1.1~1~g4f7fe52f` as long as release **v1.1** is in development and it should become `1.1 ` when released?!

## Background Information / Reproduction Steps

Hint: Debian version numbers could easily be compared with the following small utility

```
#!/bin/bash

if [ -z "$1" ]; then echo "input error"; exit; fi 
if [ -z "$2" ]; then echo "input error"; exit; fi

if $(dpkg --compare-versions "$1" "eq" "$2"); 
  then echo "$1 EQUAL $2"; exit;
  else
    if $(dpkg --compare-versions "$1" "gt" "$2"); 
      then echo "$1 GREATER than $2";
      else echo "$1 SMALLER than $2";
    fi
fi

```

## Checklist

Please provide the following information:

 - [x] open62541 Version:  (b9939bf29a0d793d5df26cddaf9f08cc2b8e3b70):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: **Debian/Linux**
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-09-29:
Yes, that's an issue...
I already got emails that launchpad is not able to upload the new package, since the version string is older than the release.

We are using `git describe --tags` to generate the version string. Therefore `v1.0-1-g4f7fe52f` makes totally sense. This means that the commit `4f7fe52f` is `1` commit after the tag `v1.0`.

`1.1~1~g4f7fe52f` is only for after-release of version `1.1`. Do you have any suggestions how we can fix the version numbering but still keeping the `git describe --tags`?

jrcoding, 2019-09-29:
As far as I understand the version numbers depend on the git tags, right?

So `git describe --tags` for the version tagged with `v1.0` gives something like `v1.0-1-g4f7fe52f`
and a version tagged with `v1.0-rc5` gives something like `v1.0-rc5-69-gb5ae7902`

If my assumption is correct then in my point of view **the tag `v1.0` should only be used if `1.0` is really released and then no more updates for `1.0` are allowed**. 
**Release in this sense means that no more versions with `1.0` as part of the version number will follow.**
Any updates needed then (e.g. solving minor  bug requests) should now increment the subversion number and the git tags should look like  `v1.1-rc1` and `v1.1-rc2` or `v1.2-rc1`and so on...

If we can do it this way (tag the versions this way) we have the following situation
e.g.

- `v1.0` is **not** relased 
  - git tag like `v1.0-rc5` or `v1.0-rc6` and so on...
  - debian package number is `1.0~rc5` or  `1.0~rc5~69~gb5ae7902`
  - debian library is `libopen62541-1` 

- `v1.0` is relased 
  - git tag is `v1.0`
  - debian package number is `1.0`
  - debian library is `libopen62541-1` 

If we now need to add PRs to `v1.0` we have to increment the subversion

- `v1.1` - update because of minor bugs
  - git tag is like `v1.1-rc1`
  - debian package number is `1.1~rc1` or `1.1~rc1~3~gdeadbeef` and so on...
  - debian library is still `libopen62541-1` 

If I made no mistake this type of versioning should it make possible to use `git describe --tags` as it is used up to now to produce version numbers but we have incrementing debian package versions each time we do a new commit?!

Please look also #3096 regarding the explanation about debian versions and how they compare

Alternative proposals...?

jrcoding, 2019-09-30:
hmm, and I made a mistake, sorry...

Even if in my opinion tags without trailing hyphen parts like `v1.0` or `v1.1` should only be given to the latest commit of that version and no version `v1.0-<...>` or `1.1-<...>` should follow my proposal above doesn't work either.

The problem is the tagged commit itself. The tagged commit itself becomes unfortunately the highest debian version because it is missing the trailing part generated by `git describe --tags` and separated by a `-`. All folowing commits will work again regarding the debian versioning.

Good advice needed here...

Pro, 2019-09-30:
Can we somehow automatically adapt the output of git describe so that the debian versioning is happy? Just adding another tag right after a release is not really practicable and it will be forgotten from time to time.

I just tested with the `--long` option, I get the same results for both versions:

On master:

```
$ git describe --tags --dirty
v1.1-dev-63-gb8115f424
$ git describe --tags --dirty --long
v1.1-dev-63-gb8115f424
```
similar for the 1.0 branch.

I think the easiest solution would be to prefix the whole debian version string with the ISO Date+Time.

Then we have something like:

`20190930T101420~1.1~dev~63~gb8115f424`

and we have strictly increasing version numbers. This should be fine since the branches only have strictly increasing changes.

jrcoding, 2019-09-30:
hmm, looks a bit ugly :-) but should work, so I don't mind!
I've no better idea at the moment...

commit f6f855d0 (2019-09-30): ci: Add datetime to debian version to ensure correct ordering (fixes #3140)
commit a1a56186 (2019-10-07): ci: Add datetime to debian version to ensure correct ordering (fixes #3140)
closed 2019-10-01 by jrcoding