open62541/open62541 #3146: UA_Connection function pointers may conflict with LWIP BSD-style socket API

jonasgreen88, 2019-10-02:
## Description
If using open62541 together with LWIP the send/recv/close function pointers of UA_Connection type gets in conflict with the LWIP BSD-style socket API if LWIP_COMPAT_SOCKETS is defined to 1.

In a project where other libs depends on the BSD-style socket API of LWIP it is not possible to build open62541 without changing the name of these function pointers.

Can we consider changing these function pointer names to get rid of this conflict?

## Background Information / Reproduction Steps
Build open62541 in a project that uses LWIP with the define LWIP_COMPAT_SOCKETS set to 1.

Used CMake options:
Use UA_ARCHITECTURE freertosLWIP and set LWIP_COMPAT_SOCKETS to 1, alternatively use your own architecture.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlgiraud, 2019-10-07:
Do you mean the defines like UA_recv in the arch headers?
This should be solveable with the arch headers.
See the comment in the lwip header file as well: https://github.com/open62541/open62541/blob/master/arch/common/ua_lwip.h

jonasgreen88, 2019-10-10:
@mlgiraud Thanks for pointing me to the comment in ua_lwip.h. I didn't notice that comment. According to that comment LWIP_COMPAT_SOCKETS must be set to 0. In my project I have it set to 1 as I have other dependencies that requires the BSD-style socket API. I guess I have to solve that somehow in my project.

mlgiraud, 2019-10-10:
Can you limit the scope where you define LWIP_COMPAT_SOCKETS as 1 and include the header to a compilation unit where you do not include open62541 stuff? that way you can use both versions.

gokhalehemanth, 2022-05-20:
> LWIP_COMPAT_SOCKETS

@jonasgreen88 were you able to resolve the conflict for socket API. I am also in same situation as you, if you could let me know how the conflict was resolved, it would be really helpful.

jonasgreen88, 2022-05-23:
Yes, define LWIP_COMPAT_SOCKETS to 2 instead of 1. Then LWIP reverse the definitions from for example `#define accept lwip_accept` to `#define lwip_accept accept`.

closed 2022-05-23 by jonasgreen88

gokhalehemanth, 2022-05-23:
Wow, coincidently I discovered the same thing today, i.e. setting LWIP_COMPAT_SOCKETS to 2 resolves the linking error. Anyways, thank you for the reply, appreciate it.