open62541/open62541 #3149: UA_DataSetFieldFlags typedefinition error (memsize = 4 instead of 2)

MartinaFelberFEM, 2019-10-03:
Hello!

I am working on a PubSub implementation. 
I tried to configure the PubSub connections, reader/writergroups etc. with a UABinaryFile. 

I can generate a UABinaryFile with another SDK's toolchain. 
Then I tried to read the binary file with open62541 UA_UABinaryFileDataType_decodeBinary function. 
The decode function works well for the connections and reader/writergroup settings.

But when I add a published dataset field the decoding stops with an error. 
I analyzed the binary file content and I think that it's correct. 

Then I looked through your type definitions and I think that the UA_DataSetFieldFlags size is wrong. 
The error occurs while parsing the DataSetMetaData/Fields/FieldFlags

The UA_DataSetFieldFlags type is instantiated correctly in the addressspace as a subtype of UInt16. The type is defined in OpcUa Spec 14 -> 6.2.2.1.4  (Release 1.04)
[image]

But in your typedefinition you force the enum type to be 4 bytes long. 

```c

/**
 * DataSetFieldFlags
 * ^^^^^^^^^^^^^^^^^
 */
typedef enum {
    UA_DATASETFIELDFLAGS_NONE = 0,
    UA_DATASETFIELDFLAGS_PROMOTEDFIELD = 1,
    __UA_DATASETFIELDFLAGS_FORCE32BIT = 0x7fffffff
} UA_DataSetFieldFlags;
UA_STATIC_ASSERT(sizeof(UA_DataSetFieldFlags) == sizeof(UA_Int32), enum_must_be_32bit);
```

Therefore the decoding of the datasetfieldflags uses an offset (memsize) of 4 bytes instead of 2. 

```c

/* DataSetFieldFlags */
{
    UA_TYPENAME("DataSetFieldFlags") /* .typeName */
    {0, UA_NODEIDTYPE_NUMERIC, {15904}}, /* .typeId */
    sizeof(UA_DataSetFieldFlags), /* .memSize */
    UA_TYPES_INT32, /* .typeIndex */
    UA_DATATYPEKIND_ENUM, /* .typeKind */
    true, /* .pointerFree */
    UA_BINARY_OVERLAYABLE_INTEGER, /* .overlayable */
    0, /* .membersSize */
    0, /* .binaryEncodingId */
    DataSetFieldFlags_members /* .members */
},
```

Then the following fields are decoded incorrectly (BuiltInType and DataType nodeId). 
And the decoding stops at parsing the DataType NodeId. Because the encoding byte of the NodeId is incorrect. 

Can you confirm this error?

Many thanks for your help!

Best regards
Martina

<br> 
Build information:

Used CMake options:

```bash
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_PUBSUB=YES -DUA_ENABLE_PUBSUB_ETH_UADP=YES -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING=YES -DUA_ENABLE_MALLOC_SINGLETON=YES -DUA_ENABLE_IMMUTABLE_NODES=YES -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=YES -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS=YES

```

I am using the current master branch version 
OS: Debian Linux
GCC 4.9.2

jpfr, 2019-10-04:
All our OPC UA enums are 32bit.

> Part 6, 5.2.4 Enumerations
Enumerations are encoded as Int32 values.
An Enumeration does not have a NULL value.

If its a 16 bit value, maybe it is not an enum after all?

MartinaFelberFEM, 2019-10-07:
Hi!
Thanks for your reply. 

I think that the DataSetFieldFlag is a bitmask with 16 Bit. 

See OpcUa spec 14 one 6.2.2.1.4 - Table 6 DataSetFieldFlags Values
The table has a column "Bit No."
The only Bit used is Bit '0' at the moment. 

[image]

One page below in Table 7 it's stated that the DataSetFieldFlags Type is a Subtype of UInt16. 

The node instantiation of the open62541 SDK is correct. The type is not a subtype of enumeration, but of UInt16.

[image]

Just the type definition in the generated C code seems to be wrong.

But there must have been other bitmask types the SDK generated correctly ...
Or maybe all OpcUa bitmasks had 32 bit size ...

jpfr, 2019-10-07:
There seems to be an inconsistency between standards.
The above citation says explicitly that all enums must be 32bit.
But the XML definitions says the length should be 16bit.

https://github.com/open62541/open62541/blob/master/tools/schema/Opc.Ua.Types.bsd#L395

Do you have access to the OPC Foundation bugtracker (Mantis)?
We should escalate this to the specification people.

MartinaFelberFEM, 2019-10-07:
But how do you handle the other opc:EnumeratedType definitions with LengthInBits != 32

-  <opc:EnumeratedType Name="NodeIdType" LengthInBits="6">
- <opc:EnumeratedType Name="AccessLevelType" LengthInBits="8" IsOptionSet="true">
- <opc:EnumeratedType Name="EventNotifierType" LengthInBits="8" IsOptionSet="true">

Or do you have a special handling for these types?

I don't have access to Mantis, but I have to ask my colleagues. I think we have a company member account. 

Do you know how I could workaround this issue in the meantime?
I don't know how the code generation works. 
 
Otherwise I would have to manually adapt the UaBinaryFile.

MartinaFelberFEM, 2019-10-18:
I have opened an issue for OpcFoundation: 
https://apps.opcfoundation.org/mantis/view.php?id=5155

jpfr, 2019-10-18:
Great! Above all the standard should be internally consistent.
Then we can make a clean implementation.

MartinaFelberFEM, 2019-10-23:
Good morning!

I got an answer from the OpcFoundation and my issue has been closed.
Their answer:
"DataSetFieldFlags is an OptionSet, not an Enumeration and thus is allowed to be 16-bits."

I haven't known the attribute **IsOptionSet="true"** and overlooked it in the .bsd file:
  `<opc:EnumeratedType Name="DataSetFieldFlags" LengthInBits="16" IsOptionSet="true">
`

Your server addressspace instantiates the DataSetFieldFlags type with the OptionSetValues. 

[image]

commit 9a795903 (2019-11-06): fix(nc): Correcting code generation for OptionSet datatypes (bitmasks)

Previously all enumerations had 32 bits size, regardless of
'IsOptionSet' or 'LengthInBits' attribute of .bsd file.
This led to incorrect type sizes for OptionSets with 'LengthInBits' != 32.
Now we use an unsigned integer type for OptionSets which is large enough
to represent the indicated 'LenghtInBits' attribute.

Fixes #3149
closed 2019-11-06 by jpfr