open62541/open62541 #3161: ERROR:  Could not open a Session with StatusCode BadCertificateUriInvalid

ENRICO-MIGLIORE, 2019-10-09:
Hello,

we are trying to connect to an OPC UA Server (Siemens 840D) using open62541-1135945 and are facing probems.

Can anybody help us out?

Thanks,
Enrico and Carmelo

PROBLEM
----
We slightly modified the client_access_control_encrypt.c file to suit our needs but can't connect to our OPC UA server. The problem is described below and we also attached some pictures, the private key and the certificate file.

SERVER
----
CNC BRAND:  SIEMENS 840D 
OPC UA Server version:  4.8
OPC UA ipv4 address and port:  192.168.1.51:4840
OPC UA Server configuration:  automatically accepts all certificates
OPC UA Server encryption:  None, Basic128Rsa15, Basic256, Basic256Sha256
OPC UA Server login:  OpcUaClient
OPC UA Server password:  SUNRISE1

CLIENT
----
PC:  Windows 10 PRO
ipv4 Address:  192.168.1.12
TOOLCHAIN:  MSVC 2019 Community Edition
PROJECT CONFIGURATION:  32 bit x86 Debug
LINKER INPUT LIBRARIES:  mbedcrypto.lib  mbedtls.lib mbedx509.lib  open62541.lib  open62541-object.lib  open62541-plugins.lib  ws2_32.lib
COMMENT:  The following libraries were built in 32 bit mode :mbedcrypto.lib  mbedtls.lib mbedx509.lib  open62541.lib  open62541-object.lib  open62541-plugins.lib
open62541 Version:  open62541-1135945

CERTIFICATES
----
CERTIFICATE :   server_cert.der 
KEY :           server_key.der
COMMENT:        You can find them in the ZIP attached file
COMMENT:        Built using create_self-signed.py with localhost.cnf (we did not modify both the script and the .cnf file)

CERTIFICATE PRESENTATION FORMAT
----
$ openssl x509 -in server_cert.der -inform der -text
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            49:fd:aa:cd:a8:f3:fd:5a:c5:99:e8:ae:57:08:07:e7:dd:fc:8e:5d
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = DE, O = open62541, CN = open62541Server@localhost
        Validity
            Not Before: Oct  9 12:46:30 2019 GMT
            Not After : Oct  8 12:46:30 2020 GMT
        Subject: C = DE, O = open62541, CN = open62541Server@localhost
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:b4:ac:72:85:cf:3c:52:f2:4a:4f:6d:81:6d:40:
                    9a:60:c6:0a:80:77:e2:dc:9f:2b:a5:25:73:d5:06:
                    14:75:7c:da:86:80:23:e6:f5:32:d8:4a:4e:60:08:
                    e3:c3:71:04:f7:19:f2:89:90:6f:6e:f7:6c:7f:a1:
                    69:e3:b9:4c:01:16:66:27:05:c3:01:b0:bd:ad:b0:
                    f3:8e:a4:8f:b0:9b:7a:6e:d6:01:96:c0:5e:5b:87:
                    2d:99:27:48:42:b5:dc:95:b1:1f:c4:e1:0e:23:50:
                    a2:b5:02:0b:2f:77:42:f7:51:c8:a9:55:e5:27:b8:
                    52:1b:a7:13:84:12:a5:02:f8:33:61:34:06:bf:ab:
                    bd:3a:05:6e:15:1f:c6:7d:39:a8:69:46:aa:95:6b:
                    4f:83:36:35:5f:31:04:b3:95:9b:8d:ab:8d:70:bb:
                    62:e3:9f:46:bd:98:8b:1e:99:4d:4e:63:ce:93:7e:
[... 173 lines cut ...]
	extNodeId.nodeId = newVariableId;

	retCode = UA_Client_addReference(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_TRUE,
		UA_STRING_NULL, extNodeId, UA_NODECLASS_VARIABLE);

	printf("addReference returned: %s\n", UA_StatusCode_name(retCode));

	retCode = UA_Client_deleteReference(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), UA_TRUE, extNodeId,
		UA_TRUE);

	printf("deleteReference returned: %s\n", UA_StatusCode_name(retCode));

	retCode = UA_Client_deleteNode(client, newVariableId, UA_TRUE);
	printf("deleteNode returned: %s\n", UA_StatusCode_name(retCode));

	/* Clean up */
	UA_VariableAttributes_clear(&newVariableAttributes);
	UA_Client_delete(client); /* Disconnects the client internally */
	return EXIT_SUCCESS;
}
[cert and key.zip](https://github.com/open62541/open62541/files/3708010/cert.and.key.zip)
[image]
[image]

Stasik0, 2019-10-12:
Does it work with uaexpert? If yes - try to take uaexpert's certificate/password and feed them into the client.

ChHaa, 2019-10-14:
Is

URI:urn:open62541.server.application

really your client URI ?

ENRICO-MIGLIORE, 2019-10-14:
> 
> 
> Is
> 
> URI:urn:open62541.server.application
> 
> really your client URI ?

thanks for yuor replay.

I open the certificate of UAexper and the field of URI is : URI:urn:SERVER:UnifiedAutomation:UaExpert.

So in my localhost.cnf I replace the original string URI.1 = urn:open62541.server.application with the new string URI.1 = urn:SERVER:UnifiedAutomation:UaExpert.

But the log output is the same:

[2019-10-14 14:57:33.692 (UTC+0200)] info/client        Connecting to endpoint opc.tcp://192.168.1.51:4840
[2019-10-14 14:57:33.692 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-10-14 14:57:33.692 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-10-14 14:57:33.692 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-10-14 14:57:33.692 (UTC+0200)] info/client        SecurityPolicy not specified -> use default #None
[2019-10-14 14:57:33.692 (UTC+0200)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2019-10-14 14:57:36.722 (UTC+0200)] info/client        TCP connection established
[2019-10-14 14:57:37.035 (UTC+0200)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-10-14 14:57:37.035 (UTC+0200)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Found 7 endpoints
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Rejecting endpoint 0: security mode doesn't match
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Rejecting endpoint 1: security mode doesn't match
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Endpoint 2 has 5 user token policies
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Selected Endpoint opc.tcp://192.168.1.51:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Selected UserTokenPolicy UserName_256_Token with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Disconnect to switch to a different SecurityPolicy
[2019-10-14 14:57:38.113 (UTC+0200)] info/client        Connecting to endpoint opc.tcp://192.168.1.51:4840
[2019-10-14 14:57:38.113 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-10-14 14:57:38.113 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-10-14 14:57:38.113 (UTC+0200)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-10-14 14:57:41.331 (UTC+0200)] info/client        TCP connection established
[2019-10-14 14:57:41.565 (UTC+0200)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-10-14 14:57:41.581 (UTC+0200)] info/client        Received a ServiceFault response with StatusCode BadCertificateUriInvalid
[2019-10-14 14:57:41.581 (UTC+0200)] error/client       Could not open a Session with StatusCode BadCertificateUriInvalid

ENRICO-MIGLIORE, 2019-10-14:
> 
> 
> Does it work with uaexpert? If yes - try to take uaexpert's certificate/password and feed them into the client.

Yes, with UaExpert work very well.

Where I can found the certificate create by Uaexpert?

I think that I found the certificate but I need also of the key.der and I don't find it!

ENRICO-MIGLIORE, 2019-10-16:
Hi,

we were finally able to read data out of our SIEMENS 840D Sinumerik OPC UA server.

As soon as we have time, we will describe what we needed to change in our project in order to get it working.

Thank you for your help,
Enrico and Carmelo

closed 2019-10-16 by ENRICO-MIGLIORE

Kforcexx, 2019-11-03:
Hi,

can you please how to change to make it work, thanks. I am face the same problem recently. thanks!

ripkop, 2019-11-26:
Hi, 
I have the same problem, could you please describe your solution, thanks

miteshec11, 2019-11-27:
The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy
So, make change in **ua_config_default.c**  file it make help...

#define APPLICATION_URI "urn:open62541.server.application" in **ua_config_default.c** File

ytc108, 2020-09-09:
> The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy
> So, make change in **ua_config_default.c** file it make help...
> 
> #define APPLICATION_URI "urn:open62541.server.application" in **ua_config_default.c** File

Hi,
thanks for the reply.
I have the same problem on Win10,then I open the file named **ua_config_default.c** ,
Below the **#define APPLICATION_URI** , there is another define that **#define APPLICATION_URI_SERVER**,
it already define the urn. as below picture show
[image]
I also define the urn in my client's code like
>cc->clientDescription.applicationUri = UA_STRING_ALLOC("urn:open62541.server.application");

I think the setting in the code should cover the pre-setting in **ua_config_default.c** ,
When I connected to server , the server say the **The client's ApplicationURI did match the certificate**,
the client show **The ServiceResult has the StatusCode BadCertificateUriInvalid**.

It seems not covered the seeting.
I already change the **localhost.cnf** in open62541/tools/certs,and my environment is mbedtls.

Can someone please describe how to change to make it work?
I will very appreciate.