open62541/open62541 #3177: Adding ws2_32.lib to all build Project (Visual Studio)

prrw, 2019-10-15:
## Description
I have built the project using cmake for Visual Studio 2015 and found out that I need the ws2_32.lib externally for my project. Would not it make sense to include it in the open62541.sln which generated by cmake for Visual Studio applications ?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):  https://github.com/open62541/open62541/commit/6574a46bd11d1c933930b2f638b9efd5ac4c10a0 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10, Miscorsoft Visual Studio 2015
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-10-15:
This already the case:
https://github.com/open62541/open62541/blob/master/arch/win32/CMakeLists.txt

How did you generate the VS build project?

prrw, 2019-10-16:
I generated it using Cmake GUI tool for Visual Studio 2015 (14). Although it was for x64 based architecture and not win32. The file link by you is i think for win32 architecture. or?

jpfr, 2019-10-22:
Win32 is just "Windows".
The definitions are also used for Windows on amd64 CPUs.

jpfr, 2020-07-07:
Please reopen if the problem persists.

closed 2025-02-13 by teccheck