open62541/open62541 #3178: UA_ENABLE_AMALGAMATION for .c and .h data generated by XML

prrw, 2019-10-15:
## Description
The header file generated from the XML using the Nodeset Compilor has the following code inside:
```
#ifdef UA_ENABLE_AMALGAMATION
# include "open62541.h"
#else
# include <open62541/server.h>
#endif
```
While compiling, since UA_ENABLE_AMALGAMATION is not defined for the .c and .h data, it always enters the else loop and asks for `<open62541/server.h>`. Hence it has to be explicitly defined before using
```
#define UA_ENABLE_AMALGAMATION
```
 But this is not the right way since, these files are generated form XML and it means this has to be editted every time it is generated. Is there any other way to solve this problem ?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=TRUE
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):  https://github.com/open62541/open62541/commit/6574a46bd11d1c933930b2f638b9efd5ac4c10a0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:  Windows 10 with Microsoft Visual Studio 2015
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

prrw, 2019-10-29:
okay. I think this problem is because of the added .c and .h file due to XML. Because we compile the XML files first and it does not know if we have used UA_ENABLE_AMALGAMATION. Thats why it assumes it as undefined.

teccheck, 2025-02-13:
The merge request was merged, so I assume this issue to be done

closed 2025-02-13 by teccheck