open62541/open62541 #3190: Performance Issue for UA-Client when starting server with a subscriber.

Fischer-Thomas, 2019-10-21:
When i try to connect with an UAClient (e.g. UAExpert) to the server with a subscriber configuration(tutorial_pubsub_subscribe) it takes multiple seconds to connect to the server. Other operations (e.g. browse) also take longer than other example programs. I also checked wireshark and saw that every response from the server(subscriber) had a delay of ~300-500ms. (other example programs ~2-5ms)

I use the default tutorial_pubsub_subscribe examples and UAExpert as client.

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-10-22:
@opcua-tsn-team-kalycito Can you check?
The subscriber socket should be set to non-blocking in order not to hog resources.

opcua-tsn-team-kalycito, 2019-10-24:
@jpfr Yes, we shall check and come up with the solution.

teccheck, 2025-02-13:
The code path has changed quite a lot since this issue was opened. It should be fixed now

closed 2025-02-13 by teccheck