open62541/open62541 #3195: how to get the access level

AndyarcherReset, 2019-10-23:
Hi,
I just create a UA client by this code, and connect to UA server, then I want to get access level for the specified node by node id, but the data is null. I just do this as below:

/************************************
 UA_ReadValueId item;
  UA_ReadValueId_init(&item);
  item.nodeId = nodeID;
  item.attributeId = UA_ATTRIBUTEID_USERACCESSLEVEL;
  UA_ReadRequest request;
  UA_ReadRequest_init(&request);
   request.nodesToRead = &item;
   request.nodesToReadSize = 1;
  UA_ReadResponse response= UA_Client_Service_read(pClient, request);
************************************/

 So how can I get the access level for this node, such as this node is "CurrentRead" or "CurrentWrite"?

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

j900155, 2019-11-21:
You need to see  the function activateSession_default and getUserAccessLevel_default.
Get user name at activateSession function then write to sessionContext.
In getUserAccessLevel funtion use sessionContext get your setting data and get nodeId.
Example in here 
https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c

AndyarcherReset, 2024-06-03:
来信已收悉，我会尽快回复，祝君安好。

teccheck, 2025-02-13:
It seems there has been no new activity for a while, so I assume this is done

closed 2025-02-13 by teccheck

AndyarcherReset, 2025-02-13:
来信已收悉，我会尽快回复，祝君安好。