open62541/open62541 #3208: Suggest add some function

fengshaoxie, 2019-10-25:
UA_Client_Subscriptions_create and UA_Client_Subscriptions_deleteSingle function can suppost  asynchronous mode？

If my connection is one thread and my subscription is another thread, I will not be able to subscribe

jpfr, 2019-10-25:
Your subscription is another thread?

Be sure not to mix up async and multi-threading.
The client lib is **not thread safe** at the moment.

That will happen eventually. But we don't have it now.

fengshaoxie, 2019-10-28:
yes,I just try to multi-threading

fengshaoxie, 2019-10-30:
It's like I'm trying to multithread and cause bugs.

[image]

fengshaoxie, 2019-11-04:
Sorry, I am not familiar with this frame. Could you tell me where to add the lock

jpfr, 2024-08-17:
This is now solved for the last release families.

closed 2024-08-17 by jpfr