open62541/open62541 #3210: Add option to disable

traversaro, 2019-10-25:
#  Description
After https://github.com/open62541/open62541/pull/1732 , the [`CMAKE_INTERPROCEDURAL_OPTIMIZATION`](https://cmake.org/cmake/help/v3.16/variable/CMAKE_INTERPROCEDURAL_OPTIMIZATION.html) global CMake option is always enabled if the compiler supports it. In gcc, this adds the `-flto` flag, that cannot be removed via `UA_ARCH_REMOVE_FLAGS` because it is added by CMake itself.
However, in some cases you may want to disable IPO even if your compilers supports it, for example if your linker has problem with it. In those cases, it would be convenient to have a `UA_ENABLE_INTERPROCEDURAL_OPTIMIZATION` option, that by default is `ON` and that would permit to disable the IPO. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue: no

jpfr, 2019-10-25:
I'm hesitant to put in a build flag for *everything*.
We should use CMake defaults and keep the build system as simple as possible.

I'd prefer either of the following
- The `CheckIPOSupported` CMake package could already provide manual overrides
- Local patching of the CMakeLists.txt

teccheck, 2025-02-13:
The build system uses `CheckIPOSupported` so this seems to be done

closed 2025-02-13 by teccheck