open62541/open62541 #3218: TCP client connect timeout not handled correctly

mtenberge, 2019-10-28:
## Description
When trying to connect to a non-reachable server, `UA_ClientConnectionTCP` in `arch/network_tcp.c` is expected to log a warning 'Trying to connect to ... timed out' and return a closed connection.

The actual behavior is, that it returns with connected=true, which is incorrect.

## Background Information / Reproduction Steps

The reason is mentioned [here](https://stackoverflow.com/questions/17769964/linux-sockets-non-blocking-connect#comment94353439_17770524): `getsockopt` does not return any error as long as the connection establishment is still in progress. In the current code, this is incorrectly interpreted as 'the connection is already established'.

### How to reproduce:
* Connect to a server, just like in the default client example code.
* Pick a server hostname/IP which is unreachable (expecting a 'connection timeout')
* The logs mention something like 'TCP connected'
* The network stack fails at a later stage (I think 'ACK receive timeout' or something) because the connection is not there.

Side effect: after the latter error occurs, the connection is terminated, but the socket file descriptor is not closed, causing a file descriptor leak (if the higher level application performs connection retries, it will eventually run out of file descriptors). I will create a separate issue for this, I'll first try to pin it down a bit more. This leak does not occur after the fix mentioned below is applied.

### Possible fix:

In `UA_ClientConnectionTCP` (`arch/network_tcp.c`), add something like this (this should probably be done somewhat cleaner, but you get the idea ;-) ):
```
+                // do a dummy read to get up-to-date errors:
+                // see https://stackoverflow.com/questions/17769964/linux-sockets-non-blocking-connect#comment94353439_17770524
+                ssize_t recverr = UA_recv(clientsockfd, NULL, 0, 0);
+
                OPTVAL_TYPE so_error;
                socklen_t len = sizeof so_error;

                int ret = UA_getsockopt(clientsockfd, SOL_SOCKET, SO_ERROR, &so_error, &len);

-                if (ret != 0 || so_error != 0) {
+                if (recverr != 0 || ret != 0 || so_error != 0) {
                    /* on connection refused we should still try to connect */
                    /* connection refused happens on localhost or local ip without timeout */
-                    if (so_error != ECONNREFUSED) {
+                    if (recverr == 0 && so_error != ECONNREFUSED) {
                        ClientNetworkLayerTCP_close(&connection);
                        UA_LOG_WARNING(logger, UA_LOGCATEGORY_NETWORK,

```

Used CMake options:

```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.16") 
-- Found Git: /usr/bin/git (found ve - [ ] Wireshark network dump attached
rsion "2.20.1") 
-- open62541 Version: 1.0.0-dirty
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- The selected architecture is: posix
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Success
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
[... 144 lines cut ...]
UA_ENABLE_TYPENAMES:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian Buster
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-07-07:
The client connection logic was revamped for 1.1.
Can you verify if this still applies?

mtenberge, 2020-07-15:
I just tested with release tag v1.1.1, with two separate target addresses:
- 1 that gives a 'no route to host'. Open62541 reports a BadDisconnect on this one,
- 1 that does not respond at all. Open62541 reports a timeout here.

So both cases seem to work correctly now. Thanks!

closed 2020-07-15 by mtenberge