open62541/open62541 #3221: TCP client socket file descriptors leaking

mtenberge, 2019-10-29:
Probably related to #3218.

## Description
In certain situations, the TCP socket file descriptor is not closed. If the application tries to (re)connect in a loop (probably with a sleep delay), the application will eventually run out of resources, as a normal application is limited to max 1024 open files.

## Background Information / Reproduction Steps
In function `UA_Client_connectTCPSecureChannel`, if the call to `client->config.connectionFunc` succeeds, but function `HelAckHandshake` fails (see #3218 for an example situation in which this can occur), `UA_Client_disconnect` is called (both in `HelAckHandshake` and in the cleanup section of `UA_Client_connectTCPSecureChannel`).

`client->connection.state` is however still set to UA_CONNECTION_OPENING, so `UA_Client_disconnect` does not call the 'client->connection.close' function.

As a result, the socket fd is not closed (as can be seen using e.g. `lsof`).

Used CMake options:
See #3218

## Logs
```
TRAC[0039] before connectWithOrWithoutLogin             
INFO[0039] Connecting to endpoint opc.tcp://10.128.49.101:4840  category="OPC-UA client"
INFO[0039] SecurityPolicy not specified -> use default #None  category="OPC-UA client"
WARN[0039] Security policy None is used to create SecureChannel. Accepting all certificates  category="OPC-UA security policy"
INFO[0039] TCP connection established                    category="OPC-UA client"
ERRO[0044] Couldn't connect the client to a TCP secure channel  category="OPC-UA client"
ERRO[0044] Receiving ACK message failed with GoodNonCriticalTimeout  category="OPC-UA network"
ERRO[0044] HEL/ACK handshake failed                      category="OPC-UA client"
<sleep delay>
TRAC[0049] before connectWithOrWithoutLogin             
WARN[0049] Security policy None is used to create SecureChannel. Accepting all certificates  category="OPC-UA security policy"
INFO[0049] Connecting to endpoint opc.tcp://10.128.49.101:4840  category="OPC-UA client"
INFO[0049] SecurityPolicy not specified -> use default #None  category="OPC-UA client"
INFO[0049] TCP connection established                    category="OPC-UA client"
ERRO[0054] Couldn't connect the client to a TCP secure channel  category="OPC-UA client"
ERRO[0054] HEL/ACK handshake failed                      category="OPC-UA client"
ERRO[0054] Receiving ACK message failed with GoodNonCriticalTimeout  category="OPC-UA network"
<etc>
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian Buster
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mtenberge, 2020-07-15:
Since #3218 is fixed, I can no longer reproduce this issue, so I can't test whether it is really fixed or the issue is still present in the code.

I leave it up to you to close this issue. Thanks!

ChrisBFX, 2022-03-10:
This (or something similar) is still an issue for us and is reproducible with the current v1.2.4 tag under linux and windows.

Steps to reproduce:
- set UA_Client_getConfig()->timeout to a value smaller than the default connect timeout of your system (f.e. 5 seconds)
- UA_Client_connect to a valid host but wrong port (no opc-ua server should be running)
- UA_Client_connect will return with BadTimeout
- call UA_Client_delete
-> socket is not closed

After digging in the source code my guess is that the problem lies in the `UA_ClientConnectionTCP_poll` function.
https://github.com/open62541/open62541/blob/master/arch/network_tcp.c#L446
`int ret = UA_select((UA_Int32)(connection->sockfd + 1), NULL, &writing_fdset,
                        &error_fdset, &tmptv);`
returns 0 after the configured timeout, which is not an error. 
`ret = UA_getsockopt(connection->sockfd, SOL_SOCKET, SO_ERROR, &so_error, &len);`
also returns no error in so_error, because the system is still trying to connect.
So the function returns `UA_STATUSCODE_GOOD`, which is passed upwards to `connectSync` in https://github.com/open62541/open62541/blob/master/src/client/ua_client_connect.c#L1096.
This function checks the timeout (again) and returns with `UA_STATUSCODE_BADTIMEOUT`. The socket is never closed in this scenario.

As a workaround we increased the timeout in the client config (we get an `UA_STATUSCODE_BADDISCONNECT` instead of Timeout as an error).

jpfr, 2022-03-15:
Fixed via #5006.

closed 2022-03-15 by jpfr