open62541/open62541 #3222: "Performing Test CC_HAS_" fails, even if compiler support the flag

renegart, 2019-10-29:
## Description
Hello,
while compiling your stack, I've recognized that ``Performing Test CC_HAS_`` tests failing when using a gcc cross-toolchain for an armv7 target.
The toolchain supports the *-fstack-protector-strong* and *-fstack-protector-strong* flags, so I checked what's wrong with the tests.

The *CMakeError.log* shows following error (don't know, why *-Werror* is set):
```bash
Run Build Command(s):/home/herrmann/mguenther/bin/hug-imx7_toolchain_current/bin/ninja cmTC_fb280 
[1/2] Building C object CMakeFiles/cmTC_fb280.dir/src.c.o
FAILED: CMakeFiles/cmTC_fb280.dir/src.c.o 
/home/herrmann/mguenther/bin/hug-imx7_toolchain_current/bin/arm-hug-linux-gnueabihf-gcc --sysroot=/home/herrmann/mguenther/bin/hug-imx7_toolchain_current/arm-hug-linux-gnueabihf/sysroot   -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g2 -DCC_HAS_-fstack-protector-strong  -Werror   -fstack-protector-strong -o CMakeFiles/cmTC_fb280.dir/src.c.o   -c src.c
<command-line>: error: ISO C99 requires whitespace after the macro name [-Werror]
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```
The issue is the **-DCC_HAS_-fstack-protector-strong** term set by ```check_c_compiler_flag``` in ``CompilerFlags.cmake``. It seems the macro definition doesn't want/allow an hyphen in its name.

A possible solution would be (https://stackoverflow.com/questions/50423280/cmake-indicates-compiler-does-not-support-flag-but-it-does)

## Background Information / Reproduction Steps

Used CMake options:

**with toolchain**
```
-- open62541 Version: 1.0.0
-- The selected architecture is: posix
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/herrmann/mguenther/git/opcua/open62541/build
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_TOOLCHAIN_FILE:FILEPATH=/home/herrmann/mguenther/bin/hug-imx7_toolchain_current/share/buildroot/toolchainfile.cmake
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED
```
**with system compiler:**
```bash
cmake -L ..
-- open62541 Version: 1.0.0
-- The selected architecture is: posix
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/herrmann/mguenther/git/opcua/open62541/build
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED
```
*CMakeOutput.log*
```bash
Run Build Command(s):/usr/bin/ninja cmTC_d49f6 
[1/2] Building C object CMakeFiles/cmTC_d49f6.dir/src.c.o
<command-line>: warning: ISO C99 requires whitespace after the macro name
[2/2] Linking C executable cmTC_d49f6
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0/current master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Gentoo with buildroot cross-toolchain for armv7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-10-29:
Fixed in #3224.

renegart, 2019-11-21:
It works on my side. Thx

teccheck, 2025-02-13:
From the comments, this looks like it's fixed

closed 2025-02-13 by teccheck