open62541/open62541 #3225: How to Update Matrix type node Values.

smccoorp, 2019-10-30:
Hello,
I want to know that how a matrix type node values can be updated. i have created a matrix type node in server by assigning some values to it. but I am not able to change the node value.
I can change the node value with UA_Server_writeValue function if node is type INT.
**How can change values of node if it is Matrix type.**

jpfr, 2019-10-30:
Try to write a Variant that contains a matrix of the appropriate size.
Variants can contain scalars, arrays, and arrays with multiple dimensions.
What type of error status code is returned?

jpfr, 2019-11-21:
Please reopen if this is not resolved.

closed 2019-11-21 by jpfr

EugenM-sie, 2020-09-18:
Hallo @jpfr, i have the same problem. I create with nodeset_compiler.py a node-object and i want to change the values of some variables in object, for example by subcribing of another server. My aim is to change the values im node. I am not sure if i can use the Varian, because i want to subscribe/request to different Information Models, which i can change dynamicly. That's why i want to xml compiler to produce Information Models und change only values. Thank you)