open62541/open62541 #3226: UA type ByteString not implemented correctly

Arno-v, 2019-10-30:
Concerns open62541 version 1.0

The bytestring datatype as used in e.g. the node id cannot contain embedded 0 values as it is handled as a ordinary '\0' terminated c string as can be seen here:
```
static UA_INLINE UA_ByteString
UA_BYTESTRING(char *chars) {
    UA_ByteString bs; bs.length = 0; bs.data = NULL;
    if(!chars)
        return bs;
    bs.length = strlen(chars); bs.data = (UA_Byte*)chars; return bs;
}

static UA_INLINE UA_ByteString
UA_BYTESTRING_ALLOC(const char *chars) {
    UA_String str = UA_String_fromChars(chars); UA_ByteString bstr;
    bstr.length = str.length; bstr.data = str.data; return bstr;
}
```
in UA_BYTESTRING_ALLOC this function is called:

```
UA_String
UA_String_fromChars(const char *src) {
    UA_String s; s.length = 0; s.data = NULL;
    if(!src)
        return s;
    s.length = strlen(src);
    if(s.length > 0) {
        s.data = (u8*)UA_malloc(s.length);
        if(!s.data) {
            s.length = 0;
            return s;
        }
        memcpy(s.data, src, s.length);
    } else {
        s.data = (u8*)UA_EMPTY_ARRAY_SENTINEL;
    }
    return s;
}
```

In these cases the strlen() function is used to determine the length of the bytearray. strlen() only works for '\0' terminated strings, see: 
[http://www.cplusplus.com/reference/cstring/strlen/](url)

It can also have the effect of creating a far larger object if the '\0' is found beyond the limit of the byte array, or it could even lead to a crash.

schroeder-, 2019-10-30:
UA_BYTESTRING_ALLOC and UA_BYTESTRING are only helper functions. You can init a UA_Bytestring your self if you want to transfer a binary data. 

```
	char* test_string = "Test\0\0ABC";

	UA_ByteString* bstr = UA_ByteString_new();
	bstr->data = (UA_Byte*) test_string;
	bstr->length = 10;

	UA_ByteString bstr = { 10, (UA_Byte*)test_string };
```

Arno-v, 2019-10-30:
I can do it myself, but in this case the helper functions do not seem very useful to me. Wouldn't it be better to create functions with the length of the byte array as an extra  parameter? I found two other helper functions which have the same issue:

- UA_NODEID_BYTESTRING
- UA_NODEID_BYTESTRING_ALLOC

jpfr, 2019-10-30:
You can create an empty bytestring of defined size via
```c
UA_StatusCode UA_ByteString_allocBuffer(UA_ByteString *bs, size_t length);
```
and the copy over the bytes.

UA_BYTESTRING_ALLOC is just a helper functions if a C-string is available.

closed 2019-11-05 by Arno-v