open62541/open62541 #3227: open62541 and windows xp

ctrotha, 2019-10-30:
## Description
running open62541 release 1.0 on Windows XP

## Background Information / Reproduction Steps
Downloaded open62541-win32.zip from open62541.org
Starting on windows xp gives the following error:

[image]

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):
 - [- ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system: WINDOWS XP
 - [ -] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ -] Wireshark network dump attached
 - [ -] Self-contained code example attached
 - [ ] Critical issue

cp-aquila, 2019-10-31:
You need a newer C runtime library.
Q quick search does not yield an installer, all XP related downloads have vanisehd from the microsoft page.

fedepell, 2019-11-24:
VS redistributable runtimes can be found here:
https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads
The newest one (2015/2017/2019) should work.

The specific issue is with:
https://github.com/open62541/open62541/blob/88dbcb6d644aaf3c38d2a1af9fc9cc2755c24c42/arch/win32/ua_architecture.h#L133
Since the secure (_s) functions are available from some version on (see also https://docs.microsoft.com/en-us/cpp/c-runtime-library/security-features-in-the-crt?view=vs-2015). 

One could also just modify that line to use the normal, non secure, snprintf to make it more compatible.

teccheck, 2025-02-04:
Since the rework of the arch specific code, this is not relevant anymore.

closed 2025-02-04 by teccheck