open62541/open62541 #3232: Question: 'Medium' nodeset build

cp-aquila, 2019-10-31:
## Description
I am developing a UA server application to be run on a Atmel SAME55 CPU. It has only limited SRAM (256k), so I have to be considerate about it.
I am using the UAModeler and the nodeset-compiler to add server-nodes.

I would like to have a *Medium* Nodeset that is as minimal as possible but still supporting the features needed for custom object instances.
With the minimal-modeset the instances of the `Gateway` nodes dont show up when running the server. With the normal nodeset that error is fixed, but I don't have enough RAM for that.

* Can I expand the minimal nodeset with the required features for this to work? 
* If yes, which features do need to be included? 
* How would I build/compile such a base nodeset?

## Background Information / Reproduction Steps
Screenshot of UAModeler:
[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version: 1.0.0
 - [x] Other OPC UA SDKs used (client or server):  UAModeler, UAExpert
 - [x] Operating system: freeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue: nope

cp-aquila, 2019-11-08:
I made some progress on that.
The real issue is the wording of the cmake options:
```# Namespace Zero
set(UA_NAMESPACE_ZERO "REDUCED" CACHE STRING "Completeness of the generated namespace zero (minimal/reduced/full)")
SET_PROPERTY(CACHE UA_NAMESPACE_ZERO PROPERTY STRINGS "MINIMAL" "REDUCED" "FULL")
if(UA_NAMESPACE_ZERO STREQUAL "MINIMAL")
    set(UA_GENERATED_NAMESPACE_ZERO OFF)
else()
    set(UA_GENERATED_NAMESPACE_ZERO ON)
endif()
```

When selecting `REDUCED` the file `Opc.Ua.NodeSet2.Minimal.xml` will get compiled and included. That is misleading, as a dumb user I would expect that the option `MINIMAL` builds the file that has minimal in its name.

I suggest to change this is a next major release. The wording could be somthing like this:
* NANO
* MINIMAL (default)
* FULL

Pro, 2019-11-08:
To create your own subset of the NodeSet2.xml, you need to modify the official NodeSet2.xml file and remove all the nodes you do not need. We did this process for our Minimal.xml.
It's hard to tell which nodes you need for your custom nodeset. It heavily depends on your use case. Therefore we can not help you on that. It's basically trial and error process.

After creating your custom NodeSet2.xml file, pass it to CMake with the option `UA_FILE_NS0`. This overrides the internal setting of the nodeset file and uses your file for initialization.

https://github.com/open62541/open62541/blob/master/CMakeLists.txt#L291

IMHO the naming of `MINIMAL`, `REDUCED`, and `FULL` is clear. The better option would be to rename the file `Opc.Ua.NodeSet2.Minimal.xml` to `Opc.Ua.NodeSet2.Reduced.xml`.

If this doesn't answer your question, please re-open the issue.

closed 2019-11-08 by Pro

cp-aquila, 2019-11-08:
> 
> IMHO the naming of `MINIMAL`, `REDUCED`, and `FULL` is clear. The better option would be to rename the file `Opc.Ua.NodeSet2.Minimal.xml` to `Opc.Ua.NodeSet2.Reduced.xml`.

Sounds better than my suggestion.

> 
> If this doesn't answer your question, please re-open the issue.

Closing this is perfectly fine.

The logical step for me is to continue working with the nodeset-compiler:
- Add more node-graph analysis stuff like show memory-usage estimation per branch
- Add options to analyze the nodeset and remove unused nodes
- Add functions to have `UA_DataTypeMember` as const memory instead of putting them on the heap (see https://github.com/open62541/open62541/issues/1806)

I will do that in seperate issues/PRs.