open62541/open62541 #3233: CI broken: Msys64 does not install

jpfr, 2019-10-31:
Hey all,

our CI pipeline doesn't complete atm.
This is the reported problem.

```
### Installing msys64 ###

Chocolatey v0.10.15
Installing the following packages:
msys2
By installing you accept licenses for the packages.

msys2 v20190524.0.0.20191030 [Approved]
msys2 package files install completed. Performing other installation steps.
Installing to: C:/msys64
Extracting 64-bit C:\ProgramData\chocolatey\lib\msys2\tools\msys2-base-x86_64-20190524.tar.xz to C:/msys64...
C:/msys64
Extracting C:\msys64\msys2-base-x86_64-20190524.tar to C:/msys64...
C:/msys64
Invoking first run to setup things like bash profile, gpg etc...
Invoking msys2 shell command: -defterm -no-start -c "ps -ef | grep '[?]' | awk '{print $2}' | xargs -r kill"

------------------ Exception ------------------

SerializedRemoteException      : 
SerializedRemoteInvocationInfo : 
ErrorRecord                    : mkdir: cannot change permissions of '/dev/shm': Permission denied
WasThrownFromThrowStatement    : False
Message                        : mkdir: cannot change permissions of '/dev/shm': Permission denied
Data                           : {}
InnerException                 : 
TargetSite                     : 
StackTrace                     : 
HelpLink                       : 
Source                         : 
HResult                        : -2146233087
```

@Pro Have you seen that before?

Pro, 2019-11-01:
See #3234 and #3235

The working builds all used older msys2. I hope pinning msys2 to that specific version fixes the CI builds again.

Pro, 2019-11-01:
See also:
https://github.com/chocolatey-community/chocolatey-coreteampackages/issues/1375

commit 4a365c47 (2019-11-01): fix(ci): Pin msys2 to specific version otherwise install fails.

Fixes #3233
commit d745c81b (2019-11-01): fix(ci): Pin msys2 to specific version otherwise install fails.

Fixes #3233
commit 894e3e43 (2019-11-01): fix(ci): Pin msys2 to specific version otherwise install fails.

Fixes #3233
PR #3235 (2019-11-02): fix(ci): Pin msys2 to specific version otherwise install fails.
Fixes #3233
closed 2019-11-02 by jpfr