open62541/open62541 #3237: UA_Server_translateBrowsePathToNodeIds, UA_Server_browseSimplifiedBrowsePath

magiino, 2019-11-02:
## Description
*UA_Server_browseSimplifiedBrowsePath* and *UA_Server_translateBrowsePathToNodeIds* returns BADNOMATCH (806F0000) in node type constructor when searching for 2 and more node ids.
When I search for single node id, it works.

## Background Information / Reproduction Steps
Project is using c++11.

**I'm just testing these brose methods so I did not wrap them into my c++ class yet.**

### UA_Server_browseSimplifiedBrowsePath() example
```
UA_STACKARRAY(UA_QualifiedName, qn, 3);
memset(qn, 0, sizeof(UA_QualifiedName) * 3);
qn[0].name = ToUA_String(ManagementNode::stateVar);
qn[0].namespaceIndex = 0;

qn[1].name = ToUA_String(ManagementNode::reservationCountVar);
qn[1].namespaceIndex = 0;

qn[2].name = ToUA_String(ManagementNode::availabilityVar);
qn[2].namespaceIndex = 0;

UA_BrowsePathResult browsePathResult1 = UA_Server_browseSimplifiedBrowsePath(server.GetServer(), instanceNodeId.GetNodeId(), 3, qn);
//returns  BADNOMATCH (806F0000)
```
### UA_Server_translateBrowsePathToNodeIds() example
```
UA_STACKARRAY(UA_RelativePathElement, rpe, 2);
memset(rpe, 0, sizeof(UA_RelativePathElement) * 2);

rpe[0].referenceTypeId = UaNodeId::HasComponent.GetNodeId();
rpe[0].isInverse = false;
rpe[0].includeSubtypes = false;
rpe[0].targetName = UA_QUALIFIEDNAME(0, "State");

rpe[1].referenceTypeId = UaNodeId::HasComponent.GetNodeId();
rpe[1].isInverse = false;
rpe[1].includeSubtypes = false;
rpe[1].targetName = UA_QUALIFIEDNAME(0, "ReservationCount");

UA_BrowsePath bp;
UA_BrowsePath_init(&bp);
bp.startingNode = instanceNodeId.GetNodeId();
bp.relativePath.elementsSize = 2;
bp.relativePath.elements = rpe;

UA_BrowsePathResult browsePathResult = UA_Server_translateBrowsePathToNodeIds(server.GetServer(), &bp);
//returns  BADNOMATCH (806F0000)
```

I checked in debug if correct values are passed to those functions.

Am I doing something wrong? Do i understand this services correctly?

## Used CMake options:

set(UA_ENABLE_HISTORIZING ON)
set(UA_ENABLE_EXPERIMENTAL_HISTORIZING ON)
set(UA_ENABLE_METHODCALLS ON)
set(UA_ENABLE_NODEMANAGEMENT ON)
set(UA_ENABLE_SUBSCRIPTIONS ON)
set(UA_ENABLE_SUBSCRIPTIONS_EVENTS ON)
set(UA_ENABLE_DISCOVERY ON)
set(UA_ENABLE_DISCOVERY_MULTICAST ON)
set(UA_ENABLE_QUERY ON)
set(UA_LOGLEVEL 100)
set(MDNSD_LOGLEVEL 100)
set(UA_NAMESPACE_ZERO FULL)
set(UA_ENABLE_STATUSCODE_DESCRIPTIONS ON)
set(UA_ENABLE_TYPEDESCRIPTION ON)

## Checklist
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0.0 - commit: b6baa5906ecae71bc15b1fd12b24d3581723fd28
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04.3 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-03:
We don’t see your information model.
The path points to nodes “deep” in the hierarchy.
Not several nodes on the same hierarchy level.

magiino, 2019-11-03:
I created this diagram (sorry it is not by spec).
The structure is simple, only one level deep.
All nodes in ManagementType are mandatory. So everything is created in constructor where I want hang ValueCallbacks for variables and MethodCallbacks to methods.

[image]

Also I have noticed, that when I hang Method callback in my constructor to any method on picture, the *objectContext* and objectId are from ObjectType and not from Object Instance I'm creating. Is it correct behavior?

```
  static UA_StatusCode MethodCallback(UA_Server *server, const UA_NodeId *sessionId,
                               void *sessionContext, const UA_NodeId *methodId,
                               void *methodContext, const UA_NodeId *objectId,
                               void *objectContext, size_t inputSize,
                               const UA_Variant *input, size_t outputSize,
                               UA_Variant *output) {
```

jpfr, 2019-11-04:
If you give a "deep" path to TranslateBrowsePathToNodeIds, then it cannot work on a flat structure.

The object context of the method call depends on the object that was used for the call.
The same method node can exist in several objects.
So you have to provide the object id for the "context" explicitly in every call request.

magiino, 2019-11-07:
> If you give a "deep" path to TranslateBrowsePathToNodeIds, then it cannot work on a flat structure.

- So you are saying that with these methods i can just search for single node id at time?

As i did in example where I serach from node "Management" for node ids of browse names "State", "ReservationCount", "Availability" => it means that I searched for single Node ID with path=Management/State/ReservationCount/Availability?

Am I correct?

> The object context of the method call depends on the object that was used for the call.
> The same method node can exist in several objects.

When I create Instance from ManagementType, the method nodes are created like on picture.
In node type constructor i search for node ids of created method nodes and I set method callback for these nodes. Then I expect as you said, that when I call this method the object context and object id will be from object instance where I call the method, but it is from ObjectType node. (objectContext is null and objectId is equal to ManagementType).

> So you have to provide the object id for the "context" explicitly in every call request.

And I do not understand what you mean with this.

jpfr, 2019-11-07:
> Am I correct?

Nearly. You search for that specific path. But there can be several nodes that are compatible with the path.

> Then I expect as you said, that when I call this method the object context and object id will be from object instance where I call the method, but it is from ObjectType node.

That sounds very unusual. The code is pretty straightforward:
https://github.com/open62541/open62541/blob/36e6e141c0fefda4e2d3528e2c48389698df8ea8/src/server/ua_services_method.c#L310

magiino, 2019-11-10:
> That sounds very unusual. The code is pretty straightforward:

I have noticed, that method nodes created in object type are copied to created instance of that type, because they have the same NodeId and thats why the objectCtx and object id in method callback are not from instance.

[image]

I m definign methods in object type with parameters:
 - Reference: HasComponent, IsMandatory
 - ctx = NULL
 - method callback = NULL

Am I defining ManagementObjectType good? Should I include method nodes with mandatory reference in any object type?

I have another questions about nodeIds of created nodes.

1. When I'm creating nodes (object, method, variable, ...), is generated `UA_NodeId *outNewNodeId` always in namespace 0? If I want to put nodes inside namespace same as parent, i have to generate node ids myself? Correct?

2. When I'm creating object node of object type in namespace 2 why are node ids of instance in namespace 0? The only way how to put them to desired namespace is to add node with `UA_Server_addNode_begin()` as in example `server_inheritance.c`?

closed 2020-05-09 by magiino