open62541/open62541 #3238: Suggest add disconnect the callback in stateCallback

fengshaoxie, 2019-11-04:
In the example-client_subscription_loop,After run the example,Then I'll unplug the cable manually,There is not callback in the function stateCallback,Just call the subscriptionInactivityCallback when timeout.I think can add the disconnect callback.

cabralfortiss, 2019-11-04:
I think I have the same issue. The scenario is as follow:
 - A clients reads a variable from a server every 1 second.
 - If the server finishes gracefully (that means, do not unplug the cable), when the clients tries to read, it detects the state change of the client which triggers a function which tries to reconnect
 - Now, if the server is unplugged (CTRL + C) and there's no gracefully disconnection,  when the client tries to read from the now-unreacheable-server, I get BadConnectionClosed but no callback on the client state, so the function to reconnect is not called.

teccheck, 2025-02-04:
Tried to reproduce. Is fixed in recent versions.

closed 2025-02-04 by teccheck