open62541/open62541 #3239: Can't find X509IdentityToken in endpoint description (use UaExpert security connect open62541 server)

Kforcexx, 2019-11-04:
## Description
I try to use uaexpert to make a security connection to open62541 server( use open62541/example/encryption/server_encryption.c). The server certificate and private key (**server_cert.der** and **server_key.der**) are created by open62541/tools/certs/create_self-signed.py.

* Start server by typing the following command as below:
> ./server_encryption **server_cert.der** **server_key.der** **uaexpert.der**

* And then use uaexpert to add server, choose security setting by Basic256sha256, server's certificate choose **server_cert.per** and the uaexpert private key choose from **~/.config/unifiedautomation/uaexpert/PKI/own/private/uaexpert_key.pem**.

* When try to connect to server, the log in uaexpert show as below
[image]

Am I do something wrong? please help. thanks!

## Background Information / Reproduction Steps
* use current newest version of master branch

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

gitsbi, 2019-12-10:
I got the exact same problem here, and in a stereotypical XKCD moment, your issue is [the only hit](https://xkcd.com/979/) google finds me. 

The server is configured with:
```
UA_ServerConfig_setBasics(&config);
UA_ServerConfig_addNetworkLayerTCP(&config, 4840, 0, 0)
UA_ServerConfig_addAllSecurityPolicies(&config, &cert, &key);
UA_ServerConfig_addAllEndpoints(&config);
```

The certificate and key are created thus: 
```
openssl req -config localhost.cnf -new -nodes -x509 -sha256 -newkey rsa:2048 -keyout localhost.key -days 365 -out localhost.crt
openssl x509 -in localhost.crt -outform der -out server_cert.der
openssl rsa -inform PEM -in localhost.key -outform PEM -out server_key.pem
```

UAExpert refuses to work with this:
```
Error: UaSessionPrivate::activateSession - can't find X509IdentityToken in endpoint description
```

Have you already found a solution to the problem?

kjeldflarup, 2020-03-13:
This works for me
```
open62541/tools/certs/create_self-signed.py .
open62541/build/bin/examples/server_encryption server_cert.der server_key.der
```

vembacher, 2022-03-14:
I tracked down the issue, the issue was that `server_config->accessControl.userTokenPolicies` and `server_config->endpoints[i]->userIdentityTokens`  did not contain the appropriate policy for `UA_USERTOKENTYPE_CERTIFICATE`. This lead to the `GetEndpointsRepsonse` not containing the certificate policy.

In Wireshark it looks the following (the second one is what you want):
[image]
[image]

I fixed it by making sure that the access control plugin code contained the correct steps to add the certificate token policy to the server (access control and endpoints). So if anyone runs into the same issue, make sure the code is present in your access control plugin, similar to the default plugin:

https://github.com/open62541/open62541/blob/9ee1ffe96dfc0c7acfd08b5a27cbc85a77e28afd/plugins/ua_accesscontrol_default.c#L272

I think the reason my plugin was missing the code was due to an error ( (unknown to me) I made while using CMake, likely related to build options.

teccheck, 2025-02-04:
Seems to be resolved.

closed 2025-02-04 by teccheck