open62541/open62541 #3242: Server's default send/receive buffer sizes rejected by Siemens S7

tboeckel, 2019-11-04:
Hi!

A customer of us has a problem with his Siemens S7-1500 contacting our open62541-based OPC UA server. The Siemens S7 has very limited hardware capabilities and cannot handle arbitrary large network packets. Hence the S7 OPC UA client demands send/receive buffers of just 8K instead of 64K (the open62541 default). Now, the problem is that the open62541 server responds the initial HEL message with its own internal default buffer sizes of 64K which are rejected on the S7 as being too big to handle. Eventually the connection is dropped by the S7 due to these incompatible buffer sizes.

For testing purposes I built an open62541-based client with limited buffer sizes (8K) and connect this one to an open62541-based server with default buffer sizes (64K). Although these two programs can establish a connection without any issues (see WireShark log), the ACK message from the server contains the server's large buffer sizes instead of possibly adapted/reduced buffer sizes to match those of the client.

My question is now: what is the correct behaviour here? Does the S7 correctly reject the ACK message with too large buffer sizes? Should open62541 reflect the S7's smaller buffer sizes in its ACK message? Or is the S7 wrong here and should not reject the larger open62541 buffer sizes? I really don't know...

The current workaround is to configure a specific send/receive buffer size on the server side to match the client's capabilities via UA_ServerConfig_setMinimalCustomBuffer() instead of just UA_ServerConfig_setMinimal(). But this looks quite awkward to me and should not be necessary. Both sides should automatically negotiate a suitable buffer size for both sides. As such open62541 should not just try/allow to enlarge its buffer sizes but also to reduce them to match a counterpart with lower capabilities. Shouldn't it?
[opcUaBro-atumZ.pcapng.gz](https://github.com/open62541/open62541/files/3803594/opcUaBro-atumZ.pcapng.gz)

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [X] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tboeckel, 2019-11-04:
There is definitely a difference between release 0.3 and release 1.0. Release 0.3 adapts the send/receive buffer sizes of the local connection configuration in processHEL(). However, release 1.0 does not perform any buffer size adaption in processHEL() or UA_Connection_processHELACK().

For me this looks like a regression between 0.3 and 1.0-

jpfr, 2019-11-04:
Definitely the buffer size should be taken from the client.
Possibly fixed by #2955.

mlang-de, 2019-11-04:
As in OPCF forum discussed (https://opcfoundation.org/forum/opc-ua-standard/client-send-maxmessagesize-smaller-then-defined-min-buffer-size-of-8192/#p1870) the Server decides which buffer size is needed and send these adjusted buffer sizes to the Client in the ACK message. The Client decides whether it can deal with these sizes. So never a bad statuscode will be send in the ACK as in open62541 v0.3.

jpfr, 2019-11-04:
Siemens is too big to ignore...
We have already put in some workarounds for other cases:

https://github.com/open62541/open62541/blob/1c718ab30a84ad0c064e099016400f2ec944abf2/src/client/ua_client_connect.c#L563
https://github.com/open62541/open62541/blob/0924ffb96d14863b8ecb4dc26b2f738319851228/plugins/ua_accesscontrol_default.c#L60

Assume that the client initializes the connection (so no reverse hello).

If the client sends a maxRecvBuffer of 4096, then the server can adjust his maxSendBuffer downwards.
This was done in the past in open62541.

If the client didn't "lie" during the HEL, then the overall result of the handshake is the same as if the server reported "ground truth" values and did the calculation after the handshake. So we can fix without changes of actual operations.

What did I miss here @mlang-de ?

mlang-de, 2019-11-05:
> If the client sends a maxRecvBuffer of 4096, then the server can adjust his maxSendBuffer downwards. This was done in the past in open62541.

With the current implementation see
https://github.com/open62541/open62541/blob/b0b902cbde26b257b6b12606d67bdd7109ebf7ea/src/ua_connection.c#L47
a maxRecvBuffer of 4096 will end in a bad statuscode.

The expected behaviour would be, that the server increase the maxRecvBuffer size to 8192 which is the minimum buffer size (see UA spec. Part 6, 1.04, chapter 6.7.1). And the client decides whether it can deal with these values.

> So we can fix without changes of actual operations.

What does this mean?

jpfr, 2019-11-05:
>  a maxRecvBuffer of 4096 will end in a bad statuscode.

The client sends ReceiveBufferSize and SendBuffersize as 8192.
The server answers with 65535 each.

We can adjust the response to match the client side.

mlang-de, 2020-01-14:
@jpfr As you already mentioned, #2955 shall handle this correct.

coconut147, 2020-02-26:
Any plans to pull #2955 soon?
Responding with larger BufferSizes to a HEL message is a violation of the UA spec...

stepman0, 2020-04-15:
Just tested this issue with version 1.0.1 of open62541 and a Siemens PLC - it worked.

tboeckel, 2020-06-15:
From my point of view this issue is solved an can be closed.

closed 2020-06-15 by tboeckel