open62541/open62541 #3243: UA_ServerConfig_setMinimal sets hardcoded username and password

stigh, 2019-11-04:
## Description

UA_ServerConfig_setMinimal calls UA_ServerConfig_setMinimalCustomBuffer, which in turn does this:

```
static const size_t usernamePasswordsSize = 2;
static UA_UsernamePasswordLogin usernamePasswords[2] = {
    {UA_STRING_STATIC("user1"), UA_STRING_STATIC("password")},
    {UA_STRING_STATIC("user2"), UA_STRING_STATIC("password1")}};

...

retval = UA_AccessControl_default(config, true,
    &config->securityPolicies[config->securityPoliciesSize-1].policyUri,
    usernamePasswordsSize, usernamePasswords);
```

This results in a logged warning
"Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network."

I would guess that the intent of using UA_ServerConfig_setMinimal would be anonymous access without any username and password. We would then get rid of the logged warning.

If the intent was to show how to configure usernames and passwords, I think it would be better to place code for this in examples.

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON .
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux / Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-04:
Well, the default config _is_ an example. But the point is still valid.

We could make the _default not add usernames.
And provide a version that takes an explicit list as an argument.

teccheck, 2025-02-04:
These default values were removed in recent version.

closed 2025-02-04 by teccheck