open62541/open62541 #3244: UA_Client_connect_username and UA_Client_connect question

fengshaoxie, 2019-11-05:
I first time use UA_Client_connect function connect to server,
And then second time use UA_Client_connect_username connect to server(not new a UA_Client and use the first time UA_Client point),
The username and password is lose efficacy ,connect failed and return 0x80200000

fengshaoxie, 2019-11-06:
use UA_Client_connect_username function connect to Prosys OPC UA Simulation Server is Error and return to 0x80020000

fredilarsen, 2020-02-02:
I get the same error, not being able to log in to the server. The server endpoint is set up with security None and user token security Basic128Rsa15.

Created a similar test client program with Eclipse Milo (Java based), and it connects immediately.

What am I missing? Is the feature missing or do I have to do something special?

jpfr, 2020-05-21:
Verify that you have the necessary encryption support.

Reopen if not solved.

closed 2020-05-21 by jpfr