open62541/open62541 #3247: subscriber receive no messages

mcHinrichs, 2019-11-05:
## Description
I try to receive messages from the _tutorial_pubsub_publisher.c_ multicast on a different pc, with the _tutorial_pubsub_subscriber.c_. On the same pc it works fine.

The wireshark dump shows that the UDP packets are received (subscriber pc):

[image]

But the subscriber doesn't show _message received:_ message:

[image]

## Background
publisher pc:     192.168.0.1
subscriber pc: 192.168.0.4

Used CMake options:

```bash
cmake -UA_ENABLE_PUBSUB=ON
```

## Checklist

Please provide the following information:

 - [x] open62541 commit: 1748f82
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [x] Operating system: ubuntu 19.04
 - [x] Logs (UA_LOGLEVEL = 100) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mcHinrichs, 2019-11-07:
I found out that the subscriber on the ubuntu vm doesn't create a PublishSubscribe object:
[image]
(subscriber on pc)

[image]
(subscriber on vm)

It's the same ubuntu on both machines. Do I have to configure something so PublishSubscribe is created or is it because of the vm?

mcHinrichs, 2019-11-25:
The problem was JSON_ENCODING was enabled. If I disable JSON_ENCODING in ccmake the PublishSubscribe object reappears.

closed 2019-11-25 by mcHinrichs

prrw, 2020-02-27:
@mcHinrichs How are you able to build using CMake having UA_ENABLE_JSON_ENCODING=ON ? In my case, if i make it on, there are errors while building.