open62541/open62541 #3249: UA_Client_connect error: double free or corruption (out)

alsheuskivictor, 2019-11-06:
If there is no connection endpoint, an error (double free or corruption) occurs during connection.

If you run the following example, then after an indefinite period of time falls with error (an error may occur in a minute, or maybe in an hour)

Error:
=============================================================
[2019-11-06 15:05:50.259 (UTC+0300)] [32minfo/client[0m	TCP connection established
[2019-11-06 15:05:50.259 (UTC+0300)] [31merror/client[0m	Reply contains the wrong service response
[2019-11-06 15:05:50.259 (UTC+0300)] [32minfo/client[0m	Error receiving the response with status code BadCommunicationError
[2019-11-06 15:05:50.259 (UTC+0300)] [32minfo/client[0m	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
*** Error in `...': double free or corruption (out): 0x00007fffffffd540 ***
=============================================================

Example
=============================================================
#include <chrono>
#include <string>
#include <thread>
#include <open62541.h>

int main () {
	std::string endpoint = "opc.tcp://localhost:48010";
	UA_Client *client {nullptr};
	client = UA_Client_new();
	UA_ClientConfig_setDefault(UA_Client_getConfig(client));
	while (1) {
		std::this_thread::sleep_for(std::chrono::milliseconds(100));
		auto status = UA_Client_connect(client, endpoint.c_str());
		if (status != UA_STATUSCODE_GOOD) {
		     continue;
		}
		UA_Client_disconnect(client);
	}
	return 0;
}
=============================================================

alsheuskivictor, 2019-11-06:
It seems the problem is solved. In function 'openSecureChannel' added 'UA_OpenSecureChannelResponse' variable initialization.

UA_StatusCode openSecureChannel(UA_Client *client, UA_Boolean renew) {
...
  UA_OpenSecureChannelResponse response;
  UA_OpenSecureChannelResponse_init(&response); // added line
... 
}

teccheck, 2025-02-04:
Seems, this is solved.

closed 2025-02-04 by teccheck