open62541/open62541 #3257: Can OPCUA PUB/SUb with UDP use  username/password and certificate?

Dc-H, 2019-11-08:
Hi all:
    I can use OPC UA pub/sub with UDP to pub and sub message successfully, but I saw the warning as follow:
[2019-11-08 06:16:45.792 (UTC+0000)] warn/server Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
     I didn't Set Configuration about username/passward,  but it seem to set it default. Then I set the username/password myself  and use UA_ServerConfig_addSecurityPolicyBasic256Sha256(config, &certificate, &privateKey). 
 But subscribers can receive message without username/password and certificate. So anyone know username/password and certificate can be used in PUB/SUB based on UDP? If they are only implemented in OPCUA C/S mode?
Thanks.

jpfr, 2019-11-08:
Username/Password is not possible for UDP-based Pub/Sub.
Because several subscribers can receive the same message.

You have two options:
- Use a different underlying protocoll for Pub/Sub. E.g. MQTT.
- Use encryption to control the set of subscribers that can decode the message.
This is an upcoming feature. Watch this space: #3255

closed 2019-11-08 by jpfr

Dc-H, 2019-11-20:
@jpfr Thanks very much. Here comes the new question. 

Has the implementation of SKS been merged into Branch:master? I didn't see the src/pubsub/ua_pubsub_sks.c. or src/pubsub/ua_pubsub_security.c in the GitHub. When can we use SKS for sign/encryption for UDP-based Pub/Sub?

By the way, can certificate and private key be used as one method to limit the subscribers for UDP-based Pub/Sub? 
Thanks.

Waner98, 2020-12-11:
Hi @Dc-H, I saw your issue because I am a front of the same problem, but I do not get any answer on my own issues. Do you know is something was added on branch master to answer your old issue concerning encryption with UDP PubSub. If not, did you find another way to counter this ? 

Thanks :)