open62541/open62541 #3258: PubSub: How to process received data ?

maleselo, 2019-11-08:
## Description

Hi open62541 community, 

I am working on a Qt application which is linked to the open62541 lib (v1.1) and I am trying  to receive data using opcua pubsub concepts.

I managed to run the publisher example (tutorial_pubsub_publisher) which is an opcua server which publish its local time every 100 ms.

I reused tutorial_pubsub_subscriber.c code into my Qt application in order to subscribe to the "DateTime" topic.

I am struggling a little bit to get my application to process the received data.  As far as I understand tutorial_pubsub_subscriber example, it creates an opcua server and a DataSet node that is updated each time the publisher send data.

And if I understand well, in my case, to be able to use this data into my application I should create an opcua client to monitor it. 

Is there any way to subscribe to process the received data directly (I mean by not using an opcua client) ?

I guess the pubsub_subscribe_standalone.c example is what I am looking for but I am not sure. And I do not understand UA_PubSubChannel concept.

So far this is what I have done:

```
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    UA_NodeId connection_identifier;
    UA_NodeId reader_group_identifier;
    UA_NodeId reader_identifier;

    UA_String transport_profile = UA_STRING("http://opcfoundation.org/UA-Profile/Transport/pubsub-udp-uadp");
    UA_NetworkAddressUrlDataType network_address_url = {UA_STRING_NULL , UA_STRING("opc.udp://224.0.0.22:4840/")};

    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setMinimal(config, 4801, nullptr);
    config->pubsubTransportLayers = (UA_PubSubTransportLayer *) UA_calloc(2, sizeof(UA_PubSubTransportLayer));
    if(!config->pubsubTransportLayers) {
        UA_Server_delete(server);
        return EXIT_FAILURE;
    }
    config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
    config->pubsubTransportLayersSize++;

    /* add pub/sub connection */
    UA_PubSubConnectionConfig connection_config;
    memset(&connection_config, 0, sizeof (connection_config));
    connection_config.name = UA_STRING("sepro_connection");
    connection_config.transportProfileUri = transport_profile;
    connection_config.enabled = UA_TRUE;
    UA_Variant_setScalar(&connection_config.address, &network_address_url, &UA_TYPES[UA_TYPES_NETWORKADDRESSURLDATATYPE]);
    connection_config.publisherId.numeric = UA_UInt32_random();
    retval |= UA_Server_addPubSubConnection(server, &connection_config, &connection_identifier);
    retval |= UA_PubSubConnection_regist(server, &connection_identifier);

    /* add reader group to the connection */
    UA_ReaderGroupConfig reader_group_config;
    memset(&reader_group_config, 0, sizeof(reader_group_config));
    reader_group_config.name = UA_STRING("sepro_reader_group");
    retval |= UA_Server_addReaderGroup(server, connection_identifier, &reader_group_config, &reader_group_identifier);

    /* add data set reader to the reader group */
    UA_DataSetReaderConfig reader_config;
    memset(&reader_config, 0, sizeof(reader_config));
    reader_config.name = UA_STRING("sepro_data_reader");
    UA_UInt16 publisher_identifier = 2234;
    reader_config.publisherId.type = &UA_TYPES[UA_TYPES_UINT16];
    reader_config.publisherId.data = &publisher_identifier;
    reader_config.writerGroupId = 100;
    reader_config.dataSetWriterId = 62541;

    /* fill data meta set */
    UA_DataSetMetaDataType_init(&reader_config.dataSetMetaData);
    reader_config.dataSetMetaData.name = UA_STRING ("sepro_data_set");
    reader_config.dataSetMetaData.fieldsSize = 1;
    reader_config.dataSetMetaData.fields = (UA_FieldMetaData *) UA_Array_new(reader_config.dataSetMetaData.fieldsSize,
                                                                             &UA_TYPES[UA_TYPES_FIELDMETADATA]);

    /* add date time data type */
    UA_FieldMetaData_init (&reader_config.dataSetMetaData.fields[0]);
    UA_NodeId_copy(&UA_TYPES[UA_TYPES_INT32].typeId, &reader_config.dataSetMetaData.fields[0].dataType);
    reader_config.dataSetMetaData.fields[0].builtInType = UA_NS0ID_DATETIME;
    reader_config.dataSetMetaData.fields[0].name = UA_STRING("date_time");
    reader_config.dataSetMetaData.fields[0].valueRank = -1;
    retval |= UA_Server_addDataSetReader(server, reader_group_identifier, &reader_config, &reader_identifier);

    /* add subscribed variables to the data set reader */
    UA_NodeId folder_id;
    UA_String folder_name = reader_config.dataSetMetaData.name;
    UA_ObjectAttributes object_attr = UA_ObjectAttributes_default;
    UA_QualifiedName folder_browse_name;
    if( folder_name.length > 0) {
        object_attr.displayName.locale = UA_STRING("en-US");
        object_attr.displayName.text = folder_name;
        folder_browse_name.namespaceIndex = 1;
        folder_browse_name.name = folder_name;
    } else {
        object_attr.displayName = UA_LOCALIZEDTEXT("en-US", "Subscribed Variables");
        folder_browse_name = UA_QUALIFIEDNAME(1, "Subscribed Variables");
    }

    UA_Server_addObjectNode(server, UA_NODEID_NULL, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), folder_browse_name,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), object_attr, nullptr, &folder_id);

    retval |= UA_Server_DataSetReader_addTargetVariables(server, &folder_id, reader_identifier, UA_PUBSUB_SDS_TARGET);
    UA_free(reader_config.dataSetMetaData.fields);

    retval |= UA_Server_run(server, &running);
    UA_Server_delete(server);
```
## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Debian 4.9.130-2
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-12:
You can set a local MonitoredItem to supervise for changes.
https://github.com/open62541/open62541/blob/5af4e6cd19db024d18c8cd041bb04817d00389d5/include/open62541/server.h#L918

Otherwise, there should be the possibility to directly add a callback to hook to the subscriber.
@opcua-tsn-team-kalycito

maleselo, 2019-11-12:
> Otherwise, there should be the possibility to directly add a callback to hook to the subscriber.

This is exactly what I'm looking for. 

I found this function into pubsub header:

https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/include/open62541/plugin/pubsub.h#L59-L61

Should I use this function to add a callback on subscriber side in order to process the received message ? Can I set the "callback" arg even if my "middleware oriented message" is an UDP layer ?

maleselo, 2019-11-15:
I managed to process data received on the subscriber side by using internal open62541 API just like it is done into pubsub_subscribe_standalone.c example. 

I had to link against static library open62541 (cmake option BUILD_SHARED_LIBS=OFF) and reused pubsub_subscribe_standalone.c example code.

I know that it is a temporary solution. Is there any release date or a roadmap for the subscriber side API ?

You can close this issue.

Regards,
Maleselo

chuongvhn, 2019-11-26:
Hi,
I'm also finding a way to subscribe to my interesting variable.
How do you specify subscribed variable your publisher is  publishing ? By Node ID of variable node ?
Anything else, like Publisher ID should be configured in subscriber ....?

maleselo, 2019-11-28:
As far as I know, the first thing you need is an OPCUA server with some data in the address space.

Then on publisher side, you need to:

- Create a PublishedDataSet and add it to your server
- Create some DataSetField and add them to your PDS

- Create a PubSub connection and add it to your server
- Create a WriterGroup and it to this connection
- Create a DataSetWriter and use the PublishedDataSet you created to publish data over the network (which can be UDP multicast, or MQTT, or AMQP). I am using UDP multicast.

This workflow is explained here: https://open62541.org/doc/current/pubsub.html

On subscriber side, I reused the code written into examples/pubsub/pubsub_subscribe_standalone.c.

In this example the function you need to look is:

```
static UA_StatusCode
subscriberListen(UA_PubSubChannel *psc)
```

This function take a PubSub channel as an argument. To receive data on subscriber side you need to configure that PubSub channel by setting the transport layer, the network address url and so on.

Then you can call the function:

```
    /* Receive messages. A regist to the message source is needed before. */
    UA_StatusCode (*receive)(UA_PubSubChannel * channel, UA_ByteString *,
                             UA_ExtensionObject *transportSettings, UA_UInt32 timeout);
```
							 
to receive data on your pubsub channel and then process them. 

As it is explained in examples/pubsub/pubsub_subscribe_standalone.c you shoud be aware that it is a temporary solution until the subscriber side API will be released

```
/**
 * IMPORTANT ANNOUNCEMENT
 * The PubSub subscriber API is currently not finished. This examples can be used to receive
 * and print the values, which are published by the tutorial_pubsub_publish example.
 * The following code uses internal API which will be later replaced by the higher-level
 * PubSub subscriber API.
 */
```

Finally I am pretty new to open62541 and OPC UA so my answer might be uncomplete.

Regards,
Maleselo

chuongvhn, 2019-11-29:
@maleselo : thanks for your explanation.  I'll try your suggestion.
For me I've just found another example how to publish and subscribe to single Uint32 variable here, it the test. I've try and it works fine for me. Now I'm investigating how to to the same thing for a data structure:  
[https://github.com/open62541/open62541/blob/5af4e6cd19db024d18c8cd041bb04817d00389d5/tests/pubsub/check_pubsub_subscribe.c#L612](https://github.com/open62541/open62541/blob/5af4e6cd19db024d18c8cd041bb04817d00389d5/tests/pubsub/check_pubsub_subscribe.c#L612)

chuongvhn, 2020-01-06:
@maleselo : Hi, I'm trying PubSub with custom data type. In Publisher's side it is able to publish the value but Subscriber's side I receive always-0. Do you have any idea ?

maleselo, 2020-01-06:
@chuongvhn I never did this. How did you create your custom data type ? Can you share the code ?

chuongvhn, 2020-01-07:
@maleselo I've started a discussion here 
https://github.com/open62541/open62541/issues/3337#issue-543829067

closed 2020-01-14 by maleselo