open62541/open62541 #3261: net_sockets.c in mbedtls gives an error that storage size of client_addr is unknown

HamidMapna, 2019-11-11:
I have a frustrating problem with installing mbedtls. I downloaded master version and generated it without any problem in cmake-gui in windows 10. I set generator as 'unix makefiles' and c compiler as ntoppc-gcc-4.4.2.exe and c++ compiler as 'ntoppc-g++-4.4.2.exe'. Anyway when I try to build it in build folder using 'make install' command, It gives me an error while building net_sockets.c source file. Error is as follow:
         

> error: storage size of 'client_addr' isn't known. 

> error: Invalid application of 'sizeof' to incomplete type 'struct sockaddr_storage'

I guess problem is with net_sockets.c and I need to set some platform specific parameters in it. I hope somebody be able to give me some hints.
[image]

HamidMapna, 2019-11-11:
Ok. I found the source of my problem. In net_sockets.c there was a define I needed to remove to let build work with windows sockets.

> #define _POSIX_C_SOURCE 200112L

closed 2019-11-11 by HamidMapna

xuchao0915170, 2021-04-13:
I had the same issue when building poco lib in QNX platform, this really worked, thanks!