open62541/open62541 #3263: ExtraPaddingSize byte of receiving chunk might be wrongly calculated

tiropp, 2019-11-11:
Dear open62541 team,

I came along a problem when trying to use the open62541 stack in a client implementation, while trying to connect to a CODESYS 3 (v3.5.14) OPC UA server. The problem only shows when following conditions are present:
  * security mode: sign or sign & encrypt
  * security policy: Basic256Sha256
  * Server side uses a certificate with key size 4096
  * Client side (open62541 based implementation) uses a key size with 2048

Under this condition when trying to open a secure channel, the server answers with a data package containing no ExtraPaddingSize. Such an answer, after decryption, can be found [here](https://paste2.org/8bxO7ID6).  

In the [decodeChunkPaddingSize()](https://github.com/open62541/open62541/blob/1.0/src/ua_securechannel.c#L1119) function where the padding size is calculated, the extra padding byte is added when the remote key length is above 2048. Which, in this scenario is the case, since the server uses a key size of 4096 bits. 

But the standard says in Table 45 (OPC UA Specification Part 6, v1.04 p. 48) that:
> ExtraPaddingSize: The most significant byte of a two-byte integer used to specify the padding size  when the key used to encrypt the message chunk is larger than 2 048 bits. This field is omitted if  the key length is less than or equal to 2 048 bits. 

Since the message was encrypted, using the  key of the client, which is 2048 bit, should the key length returned at this position in the decodeChunkPaddingSize() not using something like `getLocalKeyLength` instead of `getRemoteKeyLength`?

Also in [padChunkAsym()](https://github.com/open62541/open62541/blob/1.0/src/ua_securechannel.c#L429) the extra padding size byte is only used when the `getRemoteKeyLength` is above 2048. Out of symmetry one of the operations (`decodeChunkPaddingSize` or `padChunkAsym`) probably should use something like `getLocalKeyLength` while the other `getRemoteKeyLength`.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): Stack used by CODESYS 3 (they probably use the OPC UA foundation C stack)
 - [x] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-11:
@mlgiraud Can you have a look?

mlgiraud, 2019-11-11:
@tiropp Yes, you are correct. The decode function should use the local key size.
The padding encoding function should be correct as it is now.

mlgiraud, 2019-12-02:
@tiropp Can you confirm that #3308 fixes your problem?

tiropp, 2019-12-03:
@mlgiraud Yes, it works for me (Tested the commit [11f19885](https://github.com/open62541/open62541/pull/3308/commits/11f19885e01e9252b27961b608787ae7047d53e8) on top of 1.0)
Thanks a lot!

commit 81003a34 (2019-12-11): Fix: #3263 Padding is now calculated with the local key size during decryption.
closed 2019-12-16 by jpfr