open62541/open62541 #3265: Cannot start server_ctt executable

RavilN, 2019-11-12:
## Description

## Background Information / Reproduction Steps

Tried to run example server_ctt in Ubuntu and Windows, in both cases exits with error "No Nodestore configured in the server".

Cannot find what is required to configure it.

Used CMake options: cmake -DUA_BUILD_EXAMPLES=1 -DUA_BUILD_SELFSIGNED_CERTIFICATE=1 ../

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Commit: e6abcf76ce2b17bd97feb9abbdac61f0eef5fe40 [e6abcf7] (latest by Nov 11, 2019)
 - [ ] Other OPC UA SDKs used (client or server): Server is not running, so client does not matter.
 - [ ] Operating system: Windows 10 Pro, Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached [2019-11-11 20:03:13.228 (UTC-0700)] fatal/server       No Nodestore configured in the server
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-12:
Did you create a server `UA_Server_new()`?
That automatically sets a nodestore.

You can follow along here: https://github.com/open62541/open62541/blob/master/plugins/ua_config_default.c#L41

RavilN, 2019-11-12:
My assumption is that sample servers should run without any code change, isn't it?
I can see that ua_config_default.c has already that function UA_Server_new(), so I don't need to implement it. 
I guess it should be called by the server_ctt.c. Lokked at it now and see that it calls UA_Server_newWithConfig, and there I can see 
UA_Server *server = (UA_Server *)UA_calloc(1, sizeof(UA_Server));
So looks like UA_Server_New is not called.

In the history I see that on file server_ctt last changes are from October 3 (which is after certification, and does not have changes in server creation part) and before that - August 15, which is I believe version which was used for certification (at OPC Foundation).

So now my question is - how to build server_ctt with the same functionality and address space, what was set in certification process? I guess there must be easily reproducible way to build it? Maybe I should fetch older version?

jpfr, 2019-11-12:
The certification was performed for the 1.0 release -- which lives in the 1.0 branch of the repository.

Thanks for reporting the regression.
It is fixed here: https://github.com/open62541/open62541/pull/3271

jpfr, 2019-11-12:
Some background info.
On the master branch, the nodestore was pulled behind a plugin interface.
So the nodestore can be set in the configuration.

The server_ctt had a regression that was not automatically detected.
We build server_ctt automatically, but there are no unit tests defined against it.

RavilN, 2019-11-13:
I tried to pull the pull request, and it worked perfectly, I did not do any change myself.
Thank you for quick fix!

PR #3271 (2019-11-13): Always instantiate a nodestore if not set in the default config
Fixes #3265
commit 8d04afb0 (2019-11-13): fix(plugin): Always instantiate a nodestore if not set in the default config

Fixes #3265
closed 2019-11-13 by jpfr