open62541/open62541 #3268: Callback not called in case of error using UA_Client_readValueAttribute_async

jurafa, 2019-11-12:
## Description
Calling `UA_Client_readValueAttribute_async` while trying to read an item that is not available on the server ends only with entry in the log but the callback is never called so the client does not know that the request is no longer active.

## Background Information / Reproduction Steps
Snippet of my code:
```cpp
UA_StatusCode uaRes = UA_Client_readValueAttribute_async(m_pClient, UA_NODEID_STRING(4, "$.pidcontrol_control.CNR.ycnx"), onReadValue, pBlk, &requestId);
uaRes = UA_Client_run_iterate(m_pClient, 0);
```

The server responds correctly with `StatusCode: 0x80340000 [BadNodeIdUnknown]`.
In the file *ua_client_highlevel.c* in function `ValueAttributeRead` the code checks for 
`res->hasValue`. If the response does not have value then it just logs

```
UA_LOG_INFO(&client->config.logger, UA_LOGCATEGORY_CLIENT,
    "Cannot process the response to the async read request %u", requestId);
```
and deletes the callback without calling it. The callback is therefore never called and user is not notified about the error.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 5af4e6cd19db024d18c8cd041bb04817d00389d5
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

Wireshark dump:
[opcua.pcapng.gz](https://github.com/open62541/open62541/files/3836156/opcua.pcapng.gz)

open62541 logs:
```
Security policy None is used to create SecureChannel. Accepting all certificates
Cannot process the response to the async read request 6
```

m-drozd, 2020-06-17:
Hi, is this issue being considered in any way? I would love to use async callbacks with c++ future/promise mechanism, but not being notified of errors defeats the purpose.

teccheck, 2025-02-04:
This issue is fixed.

closed 2025-02-04 by teccheck