open62541/open62541 #3273: Cannot find documentation on how to setup certificate trust/validation [Question]

RavilN, 2019-11-13:
## Description

## Background Information / Reproduction Steps

Running example application server_ctt, generated certificate for the server using python script. Although at start-up it prints message that all client certificates will be accepted, they were not accepted until I set verification callbacks to return 0 no matter what is client's certificate.

Now, when I run CTT tests, of course some certificated validation security tests fail, because any certificate is accepted. In order for them to pass, I need somehow configure trust lists, CRLs, etc, and configure the server to apply these settings. 
Given that the server_ctt was certificaed by OPC Foundation on supporting secure connection tests, I believe it is implemented in open62541, maybe I just could not find where it is documented.

Could you please help with this?

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): latest master
 - [ ] Other OPC UA SDKs used (client or server): UA Expert, CTT latest version.
 - [ ] Operating system: Server on Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RavilN, 2019-11-19:
Hi,

Any news in this topic? Could you please provide any hints in this direction (just couple sentences would make huge difference).
I tried to pass parameters like --trustlistFolder, and others related to certificate validation, and tried to put client certificate to there (the way as it works usually with other servers, like Demo server from Unified Automation), does not help.

Thank you, Ravil

RavilN, 2019-11-19:
After further manipulations with options --trustlistFolder, --issuerlistFolder, --revocationlistFolder, now it accepts self-signed client certificate of UaExpert.
Also tried connect with the certificate issued by CA, could connect too. Did not connect without CRL, so looks like CRL is read correctly too, nice!
Did not try with more deeper level of certificate chain yet, with issuers involved.

jpfr, 2019-11-19:
You're welcome.
Please reopen if required.

closed 2019-11-19 by jpfr

kjeldflarup, 2020-01-21:
@RavilN could you write a little on what You did to make it work.