open62541/open62541 #3275: PubSub : How to publish value change ?

chuongvhn, 2019-11-14:
Hello,  

I'm  trying to understand how PubSub work. My understanding is that publisher publishes the change to OPC UA middle ware, subscriber who subscribes to publisher will be notified by OPC UA middleware.  
I'm taking a look to this example : 
https://github.com/open62541/open62541/blob/v1.0/examples/pubsub/tutorial_pubsub_publish.c
We can see that in publisher side we need to :   
```
    addPubSubConnection(server, transportProfile, networkAddressUrl);  
    addPublishedDataSet(server);  
    addDataSetField(server);  
    addWriterGroup(server);  
    addDataSetWriter(server);  
```

The published data is made inside addDataSetField, in this case is  sever time, then the server time is continuously publish across the multicast network.  
 For my case I just want to update single variable, say uint32 variable when some condition is met.
To doing so I change the the addDataSetField as following and then in another thread call Update() to update the variable.

```
UA_NodeId published_dataSet_id_, dataSet_field_id_;

static void
addDataSetField(UA_Server *server) {
    /* Add a field to the previous created PublishedDataSet */
    //UA_NodeId dataSetFieldIdent;
    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING("Server localtime");
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    dataSetFieldConfig.field.variable.publishParameters.publishedVariable =
    //UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
    UA_NODEID_NUMERIC(0, UA_NS0ID_UINT32);

    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, published_dataSet_id_,
                              &dataSetFieldConfig, &dataSet_field_id_);
}

void Update()
{
        UA_Int32 myInteger = 42;
        UA_Variant value;
        UA_Variant_setScalar(&value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
        UA_Server_writeValue(server, dataSet_field_id_, value );
}

```

I got 

> [2019-11-14 14:28:51.467 (UTC+0700)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid

MartinaFelberFEM, 2019-11-14:
Hi!
I haven't tested it, but I think your publishedVariable nodeId may be wrong.
`dataSetFieldConfig.field.variable.publishParameters.publishedVariable = UA_NODEID_NUMERIC(0, UA_NS0ID_UINT32)`

The nodeId you specified is the UINT32 type nodeId. This node is of nodeClass DataType. And nodeClass datatype has no value attribute. That's why `UA_Server_writeValue` fails with status code `BadNodeClassInvalid` I guess.

You could create a variable node with datatype int32 in the server addressspace and use it instead.

Best regards

Pro, 2019-11-15:
@chuongvhn please do not cross-post in two different places at almost the same time. Choose one place and try to get an answer there. If that doesn't work out, try another place. But please not at the same time. The community doesn't like this behavior. We do not want to have two different places to look.

See
https://stackoverflow.com/questions/58855146/opcua-open62541-pubsub-how-to-publish-value-change

chuongvhn, 2019-11-15:
I was looking around in some other examples in the repo and found the solution as below:
1. I need to add a variable node  
```
//node id
UA_NodeId gCounterNodePublisher = {1, UA_NODEIDTYPE_NUMERIC, {1234}};

addVariableNode( UA_Server * server )
{
    UA_UInt64 publishValue = 0xCAFEDEADCAFEBEAF; //0xCAFE
    UA_VariableAttributes publisherAttr = UA_VariableAttributes_default;
    UA_Variant_setScalar(&publisherAttr.value, &publishValue, &UA_TYPES[UA_TYPES_UINT64]);
    publisherAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Publisher Counter");
    publisherAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_Server_addVariableNode(server, gCounterNodePublisher,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                              UA_QUALIFIEDNAME(1, "Publisher Counter"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              publisherAttr, NULL, NULL);
}  
```

in the addDataSetField we need to configure dataset with same node ID  
```
addDataSetField( UA_Server * server )
{
    // Add a field to the previous created PublishedDataSet 
    //UA_NodeId dataSetFieldIdent;
    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING("Server localtime");
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    dataSetFieldConfig.field.variable.publishParameters.publishedVariable = 
    //UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
    //UA_NODEID_NUMERIC(0, UA_NS0ID_INT32);
    gCounterNodePublisher;
    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, published_dataSet_id_,
                              &dataSetFieldConfig, &dataSet_field_id_);
}  
```
2. Now the tricky thing is I need to call addVariableNode before addPubSubConnection  
```
    //server init and configuration
    addVariableNode(server);
    addPubSubConnection(server, transportProfile, networkAddressUrl);  
    addPublishedDataSet(server);  
    addDataSetField(server);  
    addWriterGroup(server);  
    addDataSetWriter(server);   
```

Now I can made my variable as Published data.  
3. To update the value my solution is we need to add callback function on write/read that variable. (Probably we have another way)  

```
writeVariableNode( UA_Server * server )
{
    //Write a different integer value
    static UA_UInt64 myInteger = 0xCAFEDEADCAFEBEAF;
    UA_Variant myVar;
    UA_Variant_init(&myVar);
    myInteger++;
    UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_UINT64]);
    UA_Server_writeValue(server, gCounterNodePublisher, myVar);

    // Set the status code of the value to an error code. The function
    //UA_Server_write provides access to the raw service. The above
    //UA_Server_writeValue is syntactic sugar for writing a specific node
[... 61 lines cut ...]
{
    UA_ValueCallback callback ;
    callback.onRead = onReadImpl;
    callback.onWrite = onWriteImpl;
    UA_Server_setVariableNode_valueCallback(server, gCounterNodePublisher, callback);
}  
```

The calling sequence in main function must be like this:  
```
    //variable node need to be added before pubsub connection is establish  
    addVariableNode( server_ );

    //Register callback on read/write published data
    addValueCallbackToVariableNode( server_ );
    addPubSubConnection( server_, &transportProfile, &networkAddressUrl);
    addPublishedDataSet( server_ );
    addDataSetField( server_ );
    addWriterGroup( server_ );
    addDataSetWriter( server_ );
    //addVariableNode( server_ );

    //shutdown is taken place inside run once running is false
    UA_StatusCode retval = UA_Server_run(server_, &running_);
```

closed 2019-11-15 by chuongvhn

jackybek, 2021-01-04:
hi ChuongVhn,
Did you encounter segmentation fault at this statement : 
UA_Server_addPubSubConnection(uaServer, &connectionConfig, &connectionIdentifier)

(called inside this function addPubSubConnection(server, transportProfile, networkAddressUrl))

I folllowed the tutorial but during runtime it crashed.

jackybek, 2021-08-28:
From the looks of the error message, it seemed that you are reusing an existing NodeID during creation.  I suggest to look along these lines.  Btw, I have successfully publish to mqtt and subscribe from it using port 1883.

eduardoafonsosmar, 2021-08-30:
I'm simulating a value to this existing node and I'd like to publish it. 

How could I publish that value or node via PubSub?

jackybek, 2021-08-30:
You need to implement all the pubsub related functions such as
 addPublishedDataSet(uaServer);
			addDataSetField(uaServer);
			addWriterGroup(uaServer);
			addDataSetWriter(uaServer);

for more details please refer to the tutorial on github

eduardoafonsosmar, 2021-08-30:
I'm following this sequential functions. When a variable is referenced by NodeID it doesn't works however it works with NodeID of object. It sounds a little strange!

How did you resolve your previous problem about segmentation fault in addPubSubConnection(server, transportProfile, networkAddressUrl)?

I think my problem looks like with that.

jackybek, 2021-08-30:
If i recall the previous errors were resolved with the following:
        // preallocate sendBufferSize
        connectionOptionsIndex++;       // 3
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "sendBufferSize");
        UA_UInt32 sendBufferSize = 32767;
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &sendBufferSize, &UA_TYPES[UA_TYPES_UINT32]);

        // preallocate recvBufferSize
        connectionOptionsIndex++;       // 4
        connectionOptions[connectionOptionsIndex].key = UA_QUALIFIEDNAME(namespaceIndex, "recvBufferSize");
        UA_Int32 recvBufferSize = 32767;
        UA_Variant_setScalar(&connectionOptions[connectionOptionsIndex].value, &recvBufferSize, &UA_TYPES[UA_TYPES_UINT32]);