open62541/open62541 #3280: What kind of Der format file is created by tools/cert/create_self_signed.py file?

HamidMapna, 2019-11-16:
Hi everybody.
I hope someone can advise me what kind of Der format file is created by create_self_signed.py file provided in tools/cert folder. My host architecture is PowerPc (which is a big endian arch) and have got stuck in `mbedtls_pk_parse_key(...) ` method which returns `MBEDTLS_ERR_PK_KEY_INVALID FORMAT`. I put '\0' at the end of private key char array and also I changed the `/dev/urandom` path to `/dev/random`, because there is no `urandom` file in my `/dev` folder. Within the `mbedtls_pk_parse_key(...)` method there are some Der format parser all of them are unsuccessful. Four main parsers for Der format files are `pk_parse_key_pkcs8_encrypted_der, pk_parse_key_pkcs8_unencrypted_der, pk_parse_key_pkcs1_der ` and `pk_parse_key_sec1_der`. I need to know which mentioned parser should work in terms of provided Der format private key. Anyway I am not sure if my problem is relevant to endianness of my device or maybe my private key is invalid. 
Appreciate any help.

HamidMapna, 2019-11-16:
Ok. My problem has already solved. Answer was `pk_parse_key_pkcs1_der`. But just for information to those whose device's architecture is  Bigendian. The problem with me was I had got last version of mbedtls from github. It's weird, but I noticed that the version 2.16 of [mbedtls ](https://github.com/ARMmbed/mbedtls/tree/mbedtls-2.16) just support Bigendian devices.

jpfr, 2019-11-18:
Good to see this resolved!

closed 2019-11-18 by jpfr