open62541/open62541 #3281: Executing ua_server_ctt.exe fails

quinmars, 2019-11-16:
## Description

For the [opc-ua-client](https://github.com/convertersystems/opc-ua-client) project I currently evaluate open62451's `server_ctt` to be instrumented for the [integration tests](https://github.com/convertersystems/opc-ua-client/issues/111). So far I am still struggling to get it run. All attempts were done on a windows 10 machine.

My first approach was to download the x64 zip file (V1.0). When I try to run the `ua_server_ctt.exe` a 
message box pops up stating that the `libgcc_s_sjlj-1.dll` cannot be found. I already installed `Mingw-w64` but that did not really help. I still get the same message. What package provides that dll?

My second approach was to install WSL with ubuntu and installing the ubuntu package for open62451. There I downloaded the linux 64 (V1.0) tarball. Executing the `ua_server_ctt.exe` fails with a segmentation fault:
```bash
./ua_server_ctt.exe: Symbol `UA_TYPES` has different size in shared object, consider relinking
```
The 32-bit version cannot be executed. Is this a WSL issue?

For the CI we use currently an ubuntu setup but for development we use windows. So it would be good to get it run on linux and windows. It be nice if there'd be an ubuntu package for the server_ctt so that we simply could apt-get the server_ctt.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): V1.0
 - [x] Other OPC UA SDKs used (client or server): ua_server_ctt.exe
 - [x] Operating system: Windows 10 and WSL Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-11-16:
Packages like here?
https://launchpad.net/~open62541-team/+archive/ubuntu/daily

quinmars, 2019-11-16:
I have installed `libopen62541-1` and `libopen62541-1-dev` but I do not find the `server_ctt` executable.

WontheMikeLi, 2019-11-19:
I want to try the prebuild binaries package, after extract the file, and then I go to bin folder to click "ua_server_ctt.exe" to run, but an error pop up as below
[image]

jpfr, 2019-11-19:
@WontheMikeLi Move the libopen62541.dll in the same folder als the server_ctt.
Or add the folder where it lives to the PATH environment variable.

@Pro We should have static binaries. In addition to the dll.

WontheMikeLi, 2019-11-19:
@jpfr Thanks a lot, done as instruction, another exception pop up, below is the screenshot
[image]

jpfr, 2019-11-21:
@Pro we should generate static executables for the windows package.

jpfr, 2019-11-21:
You need runtime libs from the mingle compiler. That is a mistake on our end...

quinmars, 2020-01-08:
@jpfr can you tell me in the mean time, please, what compiler exactly I have to install to get the required dll? I found out that there is the MinGW (mingw.org) and the MinGW-w64 (mingw-w65.org) compiler. Google has not provide my any useful links to the mingle compiler - probably a typo. I already have installed MinGW-w64, but there was list of different configuration to choose. The one I took didn't provide the `libgcc_s_sjlj-1.dll`.

In the long run it also would be nice if you could provide an ubuntu package. So that the server_ctt could be easily be used for the integration tests in our CI workflow.

teccheck, 2025-02-04:
The ctt exmaple does not exist in the repo anymore. This seems to be resolved.

closed 2025-02-04 by teccheck