open62541/open62541 #3285: custom namespace zero

Maerdl, 2019-11-18:
Hello,

i have just a question about a custom namespace zero.
We use the stack in an embedded device.
A new requirement came in and we have to support EngineeringUnits. Unfortunately the Full namespace zero is too big for our flash memory.

I tried to set up a custome namespace zero starting with the minimal one. 
After compiling it seems that EngineerungUnits have a lot of dependencies.
Am I right ?
Is there a "small" way to get Engineering Units into stack without blowing the code size too much ?

Thanks in advance.

jpfr, 2019-11-18:
Set `UA_ENABLE_DA` to ON in cmake.
That should set you up.

Maerdl, 2019-11-19:
Thank's for your answer, unfortunately the UA_ENABLE_DA is already set to ON.
Following features are enabled:
UA_ENABLE_DA : ON
UA_ENABLE_METHODCALLS : ON
UA_ENABLE_SUBSCRIPTIONS : ON
UA_ENABLE_TYPENAMES : ON
UA_NAMESPACE_ZERO : MINIMAL

if i try to set the Engineering Unit with
` UA_NodeId _analogItemNodeReturned;`

`status = UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "testint"),
                              outNewNodeId, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                              UA_QUALIFIEDNAME(1, "treslkf"), UA_NODEID_NUMERIC(0,UA_NS0ID_ANALOGUNITTYPE), iattr, NULL, &_analogItemNodeReturned);`

and then set the EngineeringUnit by calling 

`UA_EUInformation eu = {.namespaceUri = UA_STRING("somecompany.com"), .unitId = -1, .displayName = UA_LOCALIZEDTEXT("en-US", "m/s"), .description = UA_LOCALIZEDTEXT("en-US", "meter per second")};`

`status = UA_Server_writeObjectProperty_scalar(
            server,
            _analogItemNodeReturned,
            UA_QUALIFIEDNAME(0, "EngineeringUnits"),
            &eu,
            &UA_TYPES[UA_TYPES_EUINFORMATION]);`

UA_Server_addVariableNode fails with UA_STATUSCODE_BADTYPEDEFINITIONINVALID
and UA_Server_writeObjectProperty_scalar fails with UA_STATUSCODE_BADNODEIDUNKNOWN.

If i set the namespace to FULL everything works as expected. I'm working with the V1.0.
Any suggestions ?

Thanks!

jpfr, 2019-11-19:
Try the current master instead of 1.0.
The DA support has been revamped in a recent pull request.

Maerdl, 2019-11-19:
Thanks for your answer! 
I switched to master branch and gave it a try.
I build with "REDUCED" namespace zero.

I can set a UA_Range to a Variable with UA_NS0ID_ANALOGITEMTYPE now.

Unfortunately i can not set EngineeringUnits for Variables with type UA_NS0ID_ANALOGUNITTYPE
`UA_Server_addDataSourceVariableNode` fails in ua_services_nodemanagement.c  in function `AddNode_addRefs` at `UA_NODESTORE_GET(server, typeDefinitionId);` with `UA_STATUSCODE_BADTYPEDEFINITIONINVALID`

While starting the server the i saw that the generated `namespace0_generated.c` inits 
`/* EngineeringUnits - ns=0;i=2371 */`

May i do something wrong ?

I really appreciate your answers. 
Thanks

Pro, 2019-11-22:
You can now also use the FULL namespace zero, and remove specific nodes from the generated nodeset, including all its references.

See:
https://github.com/open62541/open62541/pull/3288

E.g. my blacklist file looks like this. If you blacklist all the nodes from this list (remove the `#`) you will get a similar nodeset as the REDUCED.

```cmake
i=79
i=83
i=94
#i=95
#i=96
i=97
i=98
i=99
i=100
i=101
i=102
i=104
i=105
i=106
i=107
i=111
i=112
i=113
i=114
i=116
i=117
i=120
i=121
i=122
i=123
i=124
i=125
i=126
i=127
i=128
i=129
i=131
#i=256
i=257
i=288
i=289
#i=290
#i=291
i=292
i=293
#i=294
#i=295
#i=296
i=297
i=298
i=299
i=300
i=301
i=302
i=303
i=304
i=305
i=306
i=307
i=308
i=309
i=310
i=311
i=312
i=313
i=314
i=315
[... 3737 lines cut ...]
ns=http://opcfoundation.org/UA/DI/;i=15058
ns=http://opcfoundation.org/UA/DI/;i=15059
ns=http://opcfoundation.org/UA/DI/;i=15060
ns=http://opcfoundation.org/UA/DI/;i=15061
ns=http://opcfoundation.org/UA/DI/;i=15062
ns=http://opcfoundation.org/UA/DI/;i=15143
ns=http://opcfoundation.org/UA/DI/;i=15292
ns=http://opcfoundation.org/UA/DI/;i=15441
ns=http://opcfoundation.org/UA/DI/;i=15590
ns=http://opcfoundation.org/UA/DI/;i=15739
ns=http://opcfoundation.org/UA/DI/;i=15891
ns=http://opcfoundation.org/UA/DI/;i=15892
ns=http://opcfoundation.org/UA/DI/;i=15893
ns=http://opcfoundation.org/UA/DI/;i=15894
ns=http://opcfoundation.org/UA/DI/;i=15897
ns=http://opcfoundation.org/UA/DI/;i=15900
ns=http://opcfoundation.org/UA/DI/;i=15901
ns=http://opcfoundation.org/UA/DI/;i=15902
ns=http://opcfoundation.org/UA/DI/;i=15903
ns=http://opcfoundation.org/UA/DI/;i=15906
ns=http://opcfoundation.org/UA/DI/;i=15909
ns=http://opcfoundation.org/UA/DI/;i=15910
ns=http://opcfoundation.org/UA/DI/;i=15911
ns=http://opcfoundation.org/UA/DI/;i=15912
```

Maerdl, 2019-12-09:
That finally solves my issue !

Thanks a lot !

closed 2019-12-09 by Maerdl