open62541/open62541 #3293: Open62541 creating .c/.h files for freeRTOS + lwIP

Student721, 2019-11-21:
Hey there,
I was trying to Install an OPC UA Server on my STM32 project with working freeRTOS + lwIP. So i downloaded/cloned the file from https://github.com/open62541/open62541 with Git. Then I followed the instruction on https://open62541.org/doc/current/building.html. Installed cmake and visual studio for windows.
Now I have started CMake GUI and enable AMALGAMATION and switched freertosLWIP on UA_ARCHITECTURE. After generated the data, I open the open62541.sln File with Visual Studio. Right click on ALL_BUILD and build. The upcoming open62541.c /.h files I copiedthe my STM folder. 

After that I have continued to follow the instructions and changed lwipopts.h. and FreeRTOSConfig.h. 
Also put the task code in my main. After build that project in STM32CubeIDE there are over 200 error in open62541.c and .h. 

What did I wrong? Any held?
Thanks to all

jpfr, 2019-11-21:
Can you post of the error messages?

Student721, 2019-11-22:
[image]

Student721, 2019-11-22:
[image]

Student721, 2019-11-22:
[image]

closed 2019-11-25 by Student721

zxiang1996, 2020-03-26:
Can you tell me how many ROM and ram are needed for the port